FreeCAD/FreeCAD #22255: CAM: Profile Operation Fails in "Individually" Mode with Multi-Face Geometries

dasarne, 2025-06-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

## Summary

When using Profile operations with `HandleMultipleFeatures = Individually` on profiles containing multiple faces, FreeCAD throws an `IndexError: list index out of range` due to failed projection in the Path.Area module.

## Environment

- **FreeCAD Version:** 1.0.1.39285 (Git)
- **OS:** Manjaro-Linux 
- **Workbench:** Path (CAM)
- **Relevant Modules:** Path.Area, PathScripts.PathUtils

## Steps to Reproduce

- Create two profiles where at least one has multiple faces (e.g., a rectangle instead of a circle)
- Select both profiles for a Profile operation
- Set `HandleMultipleFeatures` to `Individually`
- Generate toolpath → Error occurs

## Expected Behavior
Profiles should be processed sequentially without errors, with each profile milled completely before moving to the next.

## Error output

`14:53:35 <Path.Area> Area.cpp(1881): empty projection`
`14:53:35 pyException: Traceback (most recent call last):`
`File "/usr/lib/freecad/Mod/CAM/PathScripts/PathUtils.py", line 70, in new_function`
`res = function(*args, **kwargs)`
`File "/usr/lib/freecad/Mod/CAM/Path/Op/Base.py", line 788, in execute`
`result = self.opExecute(obj)`
`File "/usr/lib/freecad/Mod/CAM/Path/Op/Area.py", line 400, in opExecute`
`for shp in self.areaOpShapes(obj):`
`File "/usr/lib/freecad/Mod/CAM/Path/Op/Profile.py", line 492, in areaOpShapes`
`shapeEnv = PathUtils.getEnvelope(shape, depthparams=custDepthparams)`
`File "/usr/lib/freecad/Mod/CAM/PathScripts/PathUtils.py", line 258, in getEnvelope`
`sec = area.makeSections(heights=[0.0], project=True)[0].getShape()`
`<class 'IndexError'>: list index out of range`
`14:53:35 Profile: list index out of range`

## Technical Analysis

- **Trigger:** Multi-face geometries cause projection failures in PathUtils.getEnvelope()
- **Root Cause:** The makeSections() method returns an empty list when projecting complex faces, leading to the index error
- **Consistency:** Fails only in Individually mode; Collectively works but causes inefficient toolpaths

## Workarounds

- Use `Collectively` mode (suboptimal efficiency)
- Split multi-face profiles into separate operations
- Simplify geometries to single-face profiles to one face

## Additional Resources

- **Test-File**: [HandleMultipleFeatures-Test.zip](https://github.com/user-attachments/files/20980499/HandleMultipleFeatures-Test.zip)
- Related Issues:
  - Forum: [Path Profile on selected face fails](https://forum.freecad.org/viewtopic.php?t=76659)
  - Forum: [Contour path error](https://forum.freecad.org/viewtopic.php?style=3&t=51156)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.3, Qt 6.9.1, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: 
  * A2plus 0.4.68
  * freecad.gears 1.3.0
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PatrikRindlisbacher, 2025-07-20:
[image]

If the infeed depth is deeper than 0 it looks like this.

I have the same problem
<class 'IndexError'>: list index out of range
14:53:35 Profile: list index out of range

tarman3, 2025-07-20:
At this moment for open profile should not set depth upper or lower than shape