FreeCAD/FreeCAD #22271: FreeCAD Macro Recorded Edges Change After Parameter Modification, Causing Chamfer Failure

murray-z, 2025-07-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am using FreeCAD macro recording to capture the workflow of creating parts that involve:

Creating multiple intersecting and tangential sketches or solids.

Performing boolean operations to get the desired intersection shape.

Applying chamfer operations on specific edges of the resulting shape.

Currently, the macro records the chamfer operation like:

obj.Base = (App.ActiveDocument.Part.Shape, ["Edge1", "Edge5"])
using fixed edge indices.

However, when I modify the parameters (size, position) of the original intersecting sketches or solids, the resulting shape changes:

The edges list changes (edge count and indices).

The macro replay fails because "Edge1" and "Edge5" no longer correspond to the correct edges for chamfering.

What I Want to Achieve
I would like to:

Reuse macro-recorded logic as functions to automate part generation while modifying original parameters.

Ensure correct edges are selected for chamfering dynamically, even when the underlying geometry changes due to parameter modifications.

Avoid manual re-identification of edge indices every time parameters change.

Question
How can I reliably select the correct edges for chamfering in FreeCAD when the geometry changes, so that my macro-based automated workflows remain reusable and robust?

### Full version info

```shell
freecad version: 1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-25:
@drwho495 FYI

drwho495, 2025-10-15:
Could you try this macro on this weekly build https://github.com/drwho495/FreeCAD/releases/tag/weekly-2025.10.14

FEA-eng, 2026-01-14:
@murray-z Can you confirm if the issue is still relevant and respond to @drwho495's questions?