FreeCAD/FreeCAD #22272: Apple Developer Program License Agreement needs to be signed

cbenhagen, 2025-07-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Weekly build fail with
```
Error: HTTP status code: 403. A required agreement is missing or has expired. This request requires an in-effect agreement that has not been signed or has expired. Ensure your team has signed the necessary legal agreements and that they are not expired.
```
https://github.com/FreeCAD/FreeCAD/actions/runs/15981181616/job/45082620377

### Full version info

```shell
...
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-01:
@oursland @adrianinsaval FYI

yorikvanhavre, 2025-07-02:
Hopefully it's now solved ;)

luzpaz, 2025-07-02:
Related: https://github.com/FreeCAD/FPA/issues/255

matthiasdanner, 2025-07-12:
The certificate is also due for renewal. 
I dont know how you sign the App but likely You Need a new Distribution certificate. https://developer.apple.com/help/account/certificates/certificates-overview/

Afterwards the provisioning Profile has to be generated with the new certificate and Both Profile and certificate have to be installed on the machine which builds the app.

chennes, 2025-07-13:
It is, as usual, a many-step, multi-layered process. Now that @yorikvanhavre has agreed to the new terms, I should be able to renew the certificate.

oursland, 2025-07-24:
I think this issue can be closed.

@maxwxyz