FreeCAD/FreeCAD #22274: AppImage Crash with bus error after returning from sleep/standby/suspend

zbynekwinkler, 2025-07-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

All it says in the terminal is

```
/tmp/.mount_FreeCANmjnaG/AppRun: line 34: 2567159 Bus error               "${MAIN}" "$@"
```

The line 34 is copied to the terminal verbatim. I run AppImage based weeklies on debian with wayland.

When I tried to reproduce the problem when running under debugger, I was not successful yet. That made me think it might have something to do with the way AppImage runs so I extracted the AppImage and run the extracted `AppRun` directly. It does not crash.

I run at least one other application packaged as AppImage (freetube) which is rock solid (has not crashed once) so it is not a general AppImage problem.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/23 18:46:33
Build type: Release
Branch: main
Hash: 6faa377fbd6b7f8273f50c21a18fc004674325f9
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * A2plus 0.4.68
  * Telemetry 1.0.1
  * osh-autodoc-workbench 0.2.3
  * Defeaturing 1.2.6
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Anything else?

Writing up this issue has lead me to discovering that there is a fairly simple workaround with extracting the AppImage. But is is still unfortunate as it makes freecad look not mature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

zbynekwinkler, 2025-07-01:
Interesting. I've just tried `FreeCAD_weekly-builds-42461-conda-Linux-x86_64-py311.AppImage` and it didn't crash.

```
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.42461 (Git) AppImage
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * A2plus 0.4.68
  * Telemetry 1.0.1
  * osh-autodoc-workbench 0.2.3
  * Defeaturing 1.2.6
  * fasteners 0.5.38
```

Syres916, 2025-07-01:
@maxwxyz the weekly builds that use `Qt6.7.3` and have a `Version:` that includes `(Git shallow)` are from https://github.com/FreeCAD/FreeCAD/releases

zbynekwinkler, 2025-07-02:
I got bus error even with the appimage from the bundle repository. However, the difference being that I was able to continue working after resume from sleep and only got the bus error after many operations. With the appimage from this repository, I get bus error after any one operation right after resuming.

Dimitris75, 2025-07-03:
Do you have an Nvidia GPU?

kinghat, 2025-10-08:
same issue here:

> $ ./FreeCAD_weekly-2025.10.01-Linux-x86_64-py311.AppImage 
> FreeCAD 1.1.0, Libs: 1.1.0devR14555 (Git shallow)
> (C) 2001-2025 FreeCAD contributors
> FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
> 
> PostHog initializing. FreeCAD launch metrics are being sent.
> Telemetry Addon: Sending FreeCAD version data to Posthog
> Telemetry Addon: Sending FreeCAD system info to Posthog
> Telemetry Addon: Sending FreeCAD addon info to Posthog
> Telemetry Addon: Sending FreeCAD preferences info to Posthog
> migration.INFO: Starting CAM asset migration check
> No measure handler available for geometry of module: Measure
> /tmp/.mount_FreeCABoGned/AppRun: line 34: 102084 Bus error               (core dumped) "${MAIN}" "$@"

```
OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/157.74/1
Installed mods: 
  * Gridfinity 0.11.9
  * Cables 0.2.1
  * OpenTheme 2025.5.20
  * ThreadProfile 1.97.0
  * fasteners 0.5.42
  * FusedFilamentDesign 0.25.380
  * sheetmetal 0.7.55
  * Telemetry 1.0.2
  * woodworking 1.1.0.dev.1.1.1.2025.10.05
```

M4xTheMan, 2025-11-16:
Happens for me with the stable release (1.0.2) and the weekly-builds too. I'm on Fedora GNOME with wayland. I have not tried to run unpacked AppImages yet. I'm using a Thinkpad T14 with an integrated graphics (CPU: 11th Gen Intel Core i5-1145G7)

```
OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly3 0.12.2
  * Assembly4
  * Curves 0.6.62
  * Defeaturing 1.2.6
  * Freecad-Built-in-themes-beta 1.2.2
  * FusedFilamentDesign 0.25.200
  * Gridfinity 0.11.9
  * InventorLoader 1.5.1
  * LCInterlocking
  * Lithophane
  * Manipulator
  * OpenDark 2024.3.13
  * POV-Ray-Rendering
  * fasteners 0.5.38
  * kicadStepUpMod 11.6.9
  * lattice2 1.0.0
  * sheetmetal 0.7.22

```