FreeCAD/FreeCAD #22276: Errors in the report and file saving when working with FEM

YEAfirst, 2025-07-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

when working with FEM, there are errors in the report related to folder paths. See the example below:
12:51:12  
12:51:12  CalculiX read results...
12:51:12  Read ccx results from frd file: C:/Users/Егор Алексеевич/Documents/CALC\FEMMeshNetgen.frd
12:51:12  Read ccx results from dat file: C:/Users/Егор Алексеевич/Documents/CALC\FEMMeshNetgen.dat
12:56:17  QCoreApplication::exec: The event loop is already running
13:02:05  th. 00000078974FFE10 - Trace **D:\bld\smesh_1721382303781\work\src\SMESH\src\DriverUNV\DriverUNV_W_SMDS_Mesh.cxx** [272] : Follow exception was caught:
	**D:\bld\smesh_1721382303781\work\src\SMESH\src\DriverUNV\DriverUNV_W_SMDS_Mesh.cxx[269]::ERROR: Output file not good.**
13:02:05  Failed to auto-save document 'Unnamed'
13:11:03  th. 00000078974FFE10 - Trace **D:\bld\smesh_1721382303781\work\src\SMESH\src\DriverUNV\DriverUNV_W_SMDS_Mesh.cxx** [272] : Follow exception was caught:
	D:\bld\smesh_1721382303781\work\src\SMESH\src\DriverUNV\DriverUNV_W_SMDS_Mesh.cxx[269]::ERROR: Output file not good.
13:11:03  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: Salome Exception : D:\bld\smesh_1721382303781\work\src\SMESH\src\DriverUNV\DriverUNV_W_SMDS_Mesh.cxx[269]::ERROR: Output file not good.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-14:
@YEAfirst can you test on dev build? TIA

https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

FEA-eng, 2025-10-15:
@YEAfirst Can you confirm if the issue is still relevant in the newest weekly build?