FreeCAD/FreeCAD #22281: Draft: move ToggleDisplayMode to Core

marcuspollio, 2025-07-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Feature request:

Currently, the [Draft ToggleDisplayMode](https://wiki.freecad.org/Draft_ToggleDisplayMode/en) (shortcut <kbd>Shift</kbd> + <kbd>Spacebar</kbd>) allows one to quickly switch the Display Mode of selected objects to `Wireframe` or back to `Flat Lines`. It is only accessible via the Draft Workbench Toolbar, Menu or Context menus (on Tree or Objects).
Note that this command only applies to selected objects and changes its Display Mode property, contrary to Core `Draw Style` commands (e.g. <kbd>V</kbd>, <kbd>3</kbd>) that override the style of all objects in 3D view without changing their properties.

This tool is very handy in many use scenarios but has two unrelated issues:

- It's only available in the Draft Workbench (not even in BIM), so its discoverability is low, especially since it is packed into `Utilities`. Also, its place in this workbench is debatable: it acts on Display Mode, not very specific to drafting tools or workflow.
- Its effect is only visible when the Core Draw Style is `As is` (<kbd>V</kbd>, <kbd>1</kbd> and not overridden by one of the other Draw Style options.

To increase its use and discoverability, it is suggested to "move" this tool to the Core with its `View` siblings. The current default shortcut seems appropriate. Hopefully, the second issue listed above can be solved as well.

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS (pop:GNOME/pop/xcb)
Architecture: x86_64
Version: 1.1.0dev.42461 (Git)
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.2 shiboken 5.15.2, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.8.1
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marcuspollio, 2025-07-01:
@tetektoza You may be interested in this one as you're doing great work in this area lately 😏

maxwxyz, 2025-07-02:
@marcuspollio What is the problem of the issue?
@FreeCAD/design-working-group is this a valid feature request?

furgo16, 2025-10-15:
Still relevant

FEA-eng, 2026-01-14:
Still relevant (although it's a feature request so shouldn't become stale)

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260113 (Git shallow)
Build date: 2026/01/13 22:48:50
Build type: Release
Branch: (HEAD detached at 4cc0a1b)
Hash: 4cc0a1b718735be10261d936066cf1382dd51884
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```