FreeCAD/FreeCAD #22287: Hard crash while using measurement tool that also corrupts files

vaidastal, 2025-07-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I would like to report issue that I have been facing recently (for a few months, after release of v1), but I was unable to get logs, what happens. 

When I use FreeCAD's measure tool (at least this is highly repeatable with measure tool), sometimes, not always, FreeCAD crashes so badly that it not only closes down completely, but sometimes also ruins files including backup files. 

Due to this issue I have lost so many hours of my work and some significant projects, that I started doing complete backups of my work very often just to protect myself from FreeCAD. I never report problems about software to github, but this issue is so severe for me, that I must report.

Scenario: I am working with solid object, I select measure tool (this always seems to be fine) and when I start selecting edges for measurements, it just crashes badly. 

I have enclosed log output.

OS: Linux Debian 6.12.27

This happens on my two different laptops: one is Intel i7 12700H, and second one is AMD Ryzen 7 PRO 8840HS. I worked with FreeCAD v1.0.0 - issue is present, v1.0.1 is also present and now latest dev build is also present.
  

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.42461 (Git) AppImage
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/242.236/1
Installed mods: 
  * fasteners 0.5.38
  * Curves 0.6.63
  * kicadStepUpMod 11.7.1
  * Manipulator 1.6.0
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

Measurement

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-07-02:
I got `Selection.cpp(158)`  error when I select center of mass in measure tool. 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42461 (Git)
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38

```

[image]

vaidastal, 2025-07-03:
Hey, 

I tried this evening to play around and reproduce it, but I was not able. Not sure, what can I do to ease the fix. It really happens randomly and I have not observed any action that seems to be the cause. Sometimes I just open the project, make a quick body and with measure it just crashes, while sometimes I can do this half of the evening without a crash.

luzpaz, 2025-07-03:
> It really happens randomly and I have not observed any action that seems to be the cause.

Seem like we may have a [Heisenbug](https://en.wikipedia.org/wiki/Heisenbug).

xtemp09, 2025-07-05:
> Due to this issue I have lost so many hours of my work and some significant projects

When you launch a weekly bundle, _FeeCAD_ says: 

> WARNING: This is a development version.
> Please do not use it in a production environment.

FEA-eng, 2025-07-05:
On Windows I don't have a crash but I was going to report that the Center of mass is broken and triggers errors. Other modes seem to be ok.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42474 (Git)
Build date: 2025/07/04 04:10:37
Build type: Release
Branch: main
Hash: 927d302401caad6a9e518584e0665ae9f61509f2
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2025-09-03:
@wwmayer any chance you can take a look at this ?

wwmayer, 2025-09-03:
@vaidastal you should try a newer version.

The exception message `'App::FeaturePythonT<FeatureT>' is not a document object type` is written in Document::addObject and means that from the string `App::FeaturePythonT<FeatureT>` no valid type can be derived.

Since you are using the build with this hash 60aa5ff3730d77 here is where this exception is raised: https://github.com/FreeCAD/FreeCAD/blob/60aa5ff3730d77037ffad0c77ba96b99ef0c7df3/src/Mod/Measure/Gui/TaskMeasure.cpp#L226

The reason for this problem is that the type `Measure::MeasurePython` is an alias for `App::FeaturePythonT<MeasureBase>` and from this the system derives the string  `App::FeaturePythonT<FeatureT>` instead of `Measure::MeasurePython`.

A fix for the failure has been added with commit bb9d0a44ecb69 (from 14th July).

Reproducing the issue with the broken version is easy. All what you need to do is to measure the center of mass. This is the only mode that is affected by this bug, all other work correctly.