FreeCAD/FreeCAD #22317: Sketcher: Std_ViewFitAll in empty sketch zooms to micrometer level

Roy-043, 2025-07-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using the Std_ViewFitAll command in an empty sketch in a new file zooms to the micrometer level.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=98107

1. Open a new file.
2. Switch to the Sketcher WB.
3. Create a sketch on the XY plane.
4. Invoke Std_ViewFitAll.
5. Check the zoom level in the status bar widget (right corner).

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-07-31:
I can confirm. This happens only if in edit mode though.

If you exit edit mode, then invoking fit all does nothing.

If you enter again the sketch, then fit all does not go to micrometer, but still zoom to something like 30x20 mm or so.

PaddleStroke, 2025-07-31:
So I think that what happens is that the origin point is taken into account in the bounding box detection. And since the sketch has only this single point, then it zooms massively.

This can be confirmed if you do 'fit all' with a geometry that is not centered, far on the right for example. Then the fit all will definitely fit both the origin and the geometry. Which is not bad actually.

So I think we really just want to ignore the origin point when it is alone. Well that is not so easy to do actually. 

`View3DInventorViewer::viewAll` is the function implementing the fit all logic. it says:
 "_in the scene graph we may have objects which we want to exclude when doing a fit all. Such objects must be part of the group SoSkipBoundingGroup._"

But we have several issues here.
- The origin point is part of the general pointset. We can't put the pointset in a SoSkipBoundingGroup or all points are going to be ignored.
- Even if we could, having the origin point taken into account when doing a fitall is actually a good thing (or not?).

Interestingly the X and Y axis are not taken into account by the fit all. Even in 1.0 before those axis were of finite length. So I guess they are probably in a SoSkipBoundingGroup

PaddleStroke, 2025-07-31:
Ok so I managed to move the root point out of the point vector in coin, and put it in a SoSkipBoundingGroup.
And surprise, now it does not zoom to micrometer level, it zooms all the way to a viewport of 0mm x 0mm

[image]

Which is totally buggy.

PaddleStroke, 2025-07-31:
Ok so it's interesting. viewAll is doing a check to

```
    SbBox3f box = getBoundingBox();
    if (box.isEmpty()) {
        return;
    }
```

but this getBoundingBox is not taking into account SoSkipBoundingGroup. It's only later that the function actually skip the SoSkipBoundingGroup and so it result in a zoom to empty box.
Let's see how to fix that

furgo16, 2025-08-02:
> So I think that what happens is that the origin point is taken into account in the bounding box detection. And since the sketch has only this single point, then it zooms massively.
>
> This can be [...]

This is https://github.com/FreeCAD/FreeCAD/issues/14005, which I still think is unexpected behaviour, but others might see it differently.

PaddleStroke, 2025-08-02:
Very good, then [#22874](https://github.com/FreeCAD/FreeCAD/pull/22874) and [#22873](https://github.com/FreeCAD/FreeCAD/pull/22873) will also fix https://github.com/FreeCAD/FreeCAD/issues/14005

furgo16, 2025-08-02:
Thanks. But will it really address #14005 ? From what I read in the comments here, the PR addresses this use case:

> we really just want to ignore the origin point when it is alone.

But in #14005 the origin point is not alone, there is existing geometry, albeit unconstrained, or not constrained to an axis.

PaddleStroke, 2025-08-02:
Yes it will fix it. Fit all will focus on the rectangle only. Not on origin
and axis anymore