FreeCAD/FreeCAD #22327: Draft: importSVG.py is too slow it should be cpp

PaddleStroke, 2025-07-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

importSVG.py is too slow it should be cpp.

I'm working on a tool to import svg into sketcher. So as to not duplicate code I reused the importSVG.py. But : 
- it spams messages
- it querry about the DPI
- it is slow

The first and second are easy to solve by adding to it a 'headless' bool.

But the slowness is because it's python. It's not super slow, but it's slow enough that it is noticeable.

So I think we should translate it in cpp in Part.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wt, 2025-07-13:
I was wondering why this was so slow.

luzpaz, 2025-09-27:
@Roy-043 would you mind weighing-in ?