FreeCAD/FreeCAD #22337: New Part wb objects are not automatically added to active Part containers.

BanceFC, 2025-07-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If I have an active Part Container, when using PD, Bodies are automatically added.
In Part wb however this does not happen, Part objects are created at the root.
What then is the purpose of the active state of the Part container?

### Full version info

```shell
OS: Manjaro Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.42325 (Git) AppImage
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1
Installed mods: 
  * fasteners.backup1738584191.325799 0.5.36 (Disabled)
  * Curves 0.6.63
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Vars 0.0.2.beta1
  * dodo 1.0.1
  * A2plus 0.4.68
  * sheetmetal 0.7.23
  * Assembly4 0.50.18
  * boltsfc 2022.11.5
  * fcgear 1.0.0
  * AirPlaneDesign 0.4.1
  * frame 0.1.1
  * CurvedShapes 1.0.13
  * DynamicData 2.74.0
  * kicadStepUpMod 11.7.1
  * Plot 2024.11.26
  * Nodes 0.1.36
  * BOLTSFC 2022.11.5
  * Silk 0.2.0
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-07-05:
Can't confirm. Can your reproduce in safe mode?
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42461 (Git)
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
```

maxwxyz, 2025-07-05:
I can confirm for some commands, e.g. compound. Primitives do work.

recursiveGecko, 2025-07-05:
Related/duplicate of https://github.com/FreeCAD/FreeCAD/issues/16710 ?

maxwxyz, 2025-07-05:
yes, thanks!

closed 2025-07-05 by maxwxyz