FreeCAD/FreeCAD #22338: CAM: Sanity report has inconsistent tool metadata

z0r0, 2025-07-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As it relates to tool preparation and tool setup, I believe that a sanity report should be used to check that the configured tools and toolholders match the job in the report. Currently the following fields remain unimplemented in the tool metadata, and the tool library: 

- Manufacturer	
- Part Number	
- URL	
- Inspection Notes

I would ask that there be consistency with the following:
1. The default attributes of a particular bit.
2. The implementation of those attributes within the toolbit library editor.
3. Those default attributes being available in the sanity report if they exist for the particular toolbit in question.

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, OCC 7.9.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.7.22
  * Gridfinity 0.11.9
  * frameforge 0.1.3
```

### Subproject(s) affected?

CAM

### Anything else?

Happy to contribute a PR once there is consensus and design guidance on implementation details. 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

knipknap, 2025-07-04:
The attributes that you mention are not official FreeCAD ToolBit attributes - they come from the BTL addon. That's why they are not in the SanityCheck.

I did not propose to add the same attributes to FreeCAD main yet because I am waiting for #21722.

luzpaz, 2025-07-04:
> I did not propose to add the same attributes to FreeCAD main yet because I am waiting for https://github.com/FreeCAD/FreeCAD/issues/21722.

@knipknap should we keep this ticket open ITC or we should just take this for granted if/when your contributions move forward ?

knipknap, 2025-07-04:
IMO a separate one is not needed.

closed 2025-07-04 by luzpaz