FreeCAD/FreeCAD #22346: Freecad is slow due to many warning related to qt.accessibility.atspi

amreo, 2025-07-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Freecad is very slow due to the many warnings when I start it, and unusable after then due to the slowness.

Step to reproduce:
1) `flatpak run org.freecad.FreeCAD`

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Anything else?

Freecad log:
```
[amreo@amreo-manjarofw-pc app]$ flatpak run org.freecad.FreeCAD 
FreeCAD 1.0.1, Libs: 1.0.1R39285 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Qt: Session management error: Could not open network socket
(qt.accessibility.atspi) Could not find accessible on path: "/org/a11y/atspi/accessible/2147483682"
(qt.accessibility.atspi) Could not find accessible on path: "/org/a11y/atspi/accessible/2147483684"
(qt.accessibility.atspi) Could not find accessible on path: "/org/a11y/atspi/accessible/2147483682"
(qt.accessibility.atspi) Could not find accessible on path: "/org/a11y/atspi/accessible/2147483695"
(qt.accessibility.atspi) Accessible invalid: QAccessibleInterface(0x5622aa3aae00 invalid) "/org/a11y/atspi/accessible/2147483808"
(qt.accessibility.atspi) Accessible invalid: QAccessibleInterface(0x5622aa3e75c0 invalid) "/org/a11y/atspi/accessible/2147483809"
(qt.accessibility.atspi) Accessible invalid: QAccessibleInterface(0x5622aa3aae00 invalid) "/org/a11y/atspi/accessible/2147483808"
(qt.accessibility.atspi) Accessible invalid: QAccessibleInterface(0x5622aa3e75c0 invalid) "/org/a11y/atspi/accessible/2147483809"
(qt.accessibility.atspi) Accessible invalid: QAccessibleInterface(0x5622aa3aae00 invalid) "/org/a11y/atspi/accessible/2147483808"
QTextCursor::setPosition: Position '1141' out of range
QTextCursor::setPosition: Position '1206' out of range
QTextCursor::setPosition: Position '1271' out of range
QTextCursor::setPosition: Position '1336' out of range
QTextCursor::setPosition: Position '1401' out of range
QTextCursor::setPosition: Position '1466' out of range
QTextCursor::setPosition: Position '1531' out of range
QTextCursor::setPosition: Position '1596' out of range
QTextCursor::setPosition: Position '1661' out of range
QTextCursor::setPosition: Position '1726' out of range
QTextCursor::setPosition: Position '1791' out of range
QTextCursor::setPosition: Position '1856' out of range
QTextCursor::setPosition: Position '1921' out of range
QTextCursor::setPosition: Position '1986' out of range
QTextCursor::setPosition: Position '2051' out of range
QTextCursor::setPosition: Position '2116' out of range
QTextCursor::setPosition: Position '2181' out of range
QTextCursor::setPosition: Position '2246' out of range
QTextCursor::setPosition: Position '2311' out of range
QTextCursor::setPosition: Position '2376' out of range
QTextCursor::setPosition: Position '2441' out of range
QTextCursor::setPosition: Position '2506' out of range
QTextCursor::setPosition: Position '2571' out of range
(qt.accessibility.atspi) Could not find accessible on path: "/org/a11y/atspi/accessible/2147484112"
(qt.accessibility.atspi) Could not find accessible on path: "/org/a11y/atspi/accessible/2147484113"
QTextCursor::setPosition: Position '302' out of range
QTextCursor::setPosition: Position '302' out of range
(qt.accessibility.atspi) Could not find accessible on path: "/org/a11y/atspi/accessible/2147484113"
```

I have Gnome on manjaro-gnome

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-07-05:
I'll build the latest Releases/FreeCAD-1-0 branch on a Qt6.9.0 system to see if I can replicate the warnings but my initial gut instinct is that these are Flatpak specific.

Syres916, 2025-07-05:
Tested the self-compiled 1.0.1 build based on Qt6.9.0 as a completely new user, no config files or shared/FreeCAD folder and FreeCAD 1.0.1 started as I expected with no warnings just the standard first run `Migrating Start Workbench to Start command... done.` message. Loaded the `PartDesignExample.FCStd` file with no warnings and opened `Sketch002` error/warning free.

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.1.39291 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: e23a81e0c47f1240a562f69cf3220709f9ab0ff3
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

I'm guessing the Flatpak config for 1.0.1 is done in https://github.com/FreeCAD/org.freecad.FreeCAD/tree/master-update @adrianinsaval ?

amreo, 2025-07-05:
I think it is due to gnome. 
I have to reproduce it but I failed. Maybe it is due that I turned off the "Magnifier" accessibility feature on gnome settings. I don't know if this issue can be closed. Maybe those warning should be managed

gianlucafiore, 2025-07-06:
I don't really know what's going on, but when I open Freecad for the first time, the first line is super smooth. The second line seems to lag at about 100ms. Then it becomes unworkable. I tried restarting Freecad by default, but nothing. With version 1.0.1, it happens to a lesser extent, but the lag has been constant from the start.

OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42461 (Git) AppImage
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1

amreo, 2025-07-10:
Mhm maybe I found the right step to reproduce it:

1) Install fedora workstation 42 gnome on boxes/virtualbox/whatever you want
2) Install freecad from flatpak on it
3) Open gnome settings --> accessibility --> (screen) magnifier and enable magnifier
4) Open freecad with `flatpak run org.freecad.FreeCAD`

amreo, 2026-02-19:
> [@amreo](https://github.com/amreo) could you confirm that it works on appimage?

Yes, Freecad works correctly and I can't reproduce the issue on the latest appimage.

```
OS: Manjaro Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow) AppImage
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
```

gucio321, 2026-02-19:
And note that "standalone" (non-appimage) build even from latest weekly, still doesn't work...

amreo, 2026-02-19:
@maxwxyz was it verified on flatpak?

FEA-eng, 2026-08-20:
> I don't have tried on the build from source. I will try it

Can you check if it's still relevant in official or custom builds?

amreo, 2026-08-21:
> Can you check if it's still relevant in official or custom builds?
Yes:

The bug is reproducible with:
* Flatpak 1.1.3
```
OS: KDE Flatpak runtime (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.3.44987 (Git)
Build date: 2026/07/25 04:52:02
Build type: Release
Branch: (HEAD detached at 1.1.3)
Hash: 145529fe741292ff0b3977a01195bf0247425794
Python 3.13.13, Qt 6.11.1, Coin 4.0.8, Vtk 9.4.2, boost 1_86, Eigen3 3.4.1, PySide 6.11.1
shiboken 6.11.1, SMESH 7.7.1.0, xerces-c 3.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.372/1
```
* Local build, 2026-08-20:
```
OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.2.0dev.20260519 (Git shallow)
Build date: 2026/05/19 20:59:45
Build type: Unknown
Branch: (HEAD detached at 5a5d8d042)
Hash: 5a5d8d0425fdb440fb5dcb47f1c48422d07d3267
Python 3.13.13, Qt 6.10.2, Coin 4.0.7, Vtk 9.2.6, boost 1_83, Eigen3 3.4.0, PySide 6.10.2
shiboken 6.10.2, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 , OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/143.372/1
```
 
The bug is not reproducible with:
* Freecad weekly 2026-08-20:
```
OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 26.3.0dev.20260819 (Git shallow) AppImage
Build date: 2026/08/19 23:20:55
Build type: Release
Branch: (HEAD detached at 2bb00c918)
Hash: 2bb00c91865d376c4ab0441a3a887b4f2d0a3090
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/143.372/1
OpenGL version/vendor/renderer: 4.5 (Compatibility Profile) Mesa 25.1.9/Mesa/llvmpipe (LLVM 20.1.8, 256 bits)
```

Robinlovelace, 2026-09-20:
Adding a data point: on 1.1.3 flatpak (Ubuntu 26.04, GNOME/Wayland → XWayland, NVIDIA) this same log spam escalates to a full **startup deadlock**, not just slowness.

- The window is mapped, but the main thread sits in `futex_do_wait` at 0% CPU and never answers `_NET_WM_PING`, so GNOME offers Wait / Force Quit. Every launch repeated it (6/6).
- Confirming the reports above: `--env=QT_ACCESSIBILITY=0 --env=NO_AT_BRIDGE=1` changes nothing.
- What does fix it: `flatpak run --no-a11y-bus` — 6/6 launches hung without the flag, 3/3 started clean with it.
- Other candidates all still hung, so it isn't the GPU or fonts: `LIBGL_ALWAYS_SOFTWARE=1`, `QT_QPA_PLATFORMTHEME=`, `--no-session-bus`. (`QT_QPA_PLATFORM=wayland` gives no window at all — Coin3D needs xcb.)

Possibly related to the AT-SPI broadcast-filtering bug in xdg-dbus-proxy 0.1.6/0.1.7 (GHSA-r7hp-698j-2h6c, fixed in 0.1.8) plus the matching flatpak fix (commit 2afb4cf, flatpak ≥ 1.18.1) — my stack is flatpak 1.16.6 + xdg-dbus-proxy 0.1.7, so I can't yet tell that apart from a Qt-side cause. Worth a test by anyone on the newer flatpak.

One packaging note: this can't be baked into the app's own metadata — `flatpak build-finish` has no a11y-bus toggle (only `--a11y-own-name`) — so the workaround has to live in the launcher/desktop entry.

amreo, 2026-09-21:
I'm not sure that the bug can be solved at the launcher/desktop entry, because it also applies with a local build

amreo, 2026-09-21:
@maxwxyz can you remove flatpak label?