FreeCAD/FreeCAD #22352: Sketcher: Crash while using trim tool

lagnat, 2025-07-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This does not appear to be a regression of https://github.com/FreeCAD/FreeCAD/issues/21013 and unfortunately I'm not able to refine the crash down to a simple example.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [25822]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-07-05 12:25:46.3038 -0400
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        80057660-6D13-007C-563C-EE545423E9F4

Sleep/Wake UUID:       BCB30292-DD54-4542-A587-AA5D4E32B510

Time Awake Since Boot: 2600000 seconds
Time Since Wake:       1225 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [25822]

VM Region Info: 0 is not in any region.  Bytes before following region: 4376641536
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104de4000-104e04000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Part.so                       	       0x12aad51f0 Part::GeomCurve::intersect(Part::GeomCurve const*, std::__1::vector<std::__1::pair<Base::Vector3<double>, Base::Vector3<double>>, std::__1::allocator<std::__1::pair<Base::Vector3<double>, Base::Vector3<double>>>>&, double) const + 48
1   Sketcher.so                   	       0x16d41101c Sketcher::SketchObject::deriveConstraintsForPieces(int, std::__1::vector<int, std::__1::allocator<int>> const&, std::__1::vector<Part::Geometry const*, std::__1::allocator<Part::Geometry const*>> const&, Sketcher::Constraint const*, std::__1::vector<Sketcher::Constraint*, std::__1::allocator<Sketcher::Constraint*>>&) const + 1972
2   Sketcher.so                   	       0x16d40ef08 Sketcher::SketchObject::deriveConstraintsForPieces(int, std::__1::vector<int, std::__1::allocator<int>> const&, Sketcher::Constraint const*, std::__1::vector<Sketcher::Constraint*, std::__1::allocator<Sketcher::Constraint*>>&) const + 616
3   Sketcher.so                   	       0x16d40e684 createNewConstraintsForTrim(Sketcher::SketchObject const*, int, std::__1::array<int, 2ul> const&, std::__1::array<Base::Vector3<double>, 2ul> const&, std::__1::vector<int, std::__1::allocator<int>> const&, std::__1::vector<int, std::__1::allocator<int>>&, std::__1::vector<Sketcher::Constraint*, std::__1::allocator<Sketcher::Constraint*>>&, std::__1::set<int, std::__1::greater<void>, std::__1::allocator<int>>&) + 792
4   Sketcher.so                   	       0x16d40f3ec Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) + 1112
5   Sketcher.so                   	       0x16d517d84 Sketcher::SketchObjectPy::trim(_object*) + 136
6   Sketcher.so                   	       0x16d508960 Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) + 128
7   libpython3.11.dylib           	       0x10563a120 cfunction_call + 284
8   libpython3.11.dylib           	       0x1055e2394 _PyObject_MakeTpCall + 332
9   libpython3.11.dylib           	       0x1056e5b74 _PyEval_EvalFrameDefault + 45376
10  libpython3.11.dylib           	       0x1056d9a58 PyEval_EvalCode + 204
11  libpython3.11.dylib           	       0x10574006c run_mod + 144
12  libpython3.11.dylib           	       0x105743d5c PyRun_StringFlags + 112
13  libFreeCADBase.dylib          	       0x105b864c8 Base::InterpreterSingleton::runString(char const*) + 108
14  libFreeCADGui.dylib           	       0x10774a698 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 168
15  libFreeCADGui.dylib           	       0x10774a52c Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) + 120
16  SketcherGui.so                	       0x16e842b84 void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, int&, double&, double&) + 280
17  SketcherGui.so                	       0x16e842960 SketcherGui::DrawSketchHandlerTrimming::executeCommands(Base::Vector2d) + 324
18  SketcherGui.so                	       0x16e84248c SketcherGui::DrawSketchHandlerTrimming::releaseButton(Base::Vector2d) + 16
19  SketcherGui.so                	       0x16e9cfe74 SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) + 1920
20  libFreeCADGui.dylib           	       0x107c6aef4 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 544
[... 57 lines cut ...]
```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42461 (Git)
Build date: 2025/06/30 16:12:02
Build type: Release
Branch: main
Hash: 60aa5ff3730d77037ffad0c77ba96b99ef0c7df3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-05:
@lagnat can you upload the example file ? (append .zip to the filename so it can attach)

lagnat, 2025-07-05:
[Crash.FCStd.zip](https://github.com/user-attachments/files/21078671/Crash.FCStd.zip)

floofywolf, 2025-07-06:
I also have a crash using the trim tool, that may be the same crash.  

Save file and reproduction instructions:
[freecadcrash01.FCStd.zip](https://github.com/user-attachments/files/21085619/freecadcrash01.FCStd.zip)
[image]

Depending on the order I trim segments in, the following may happen:
It segfaults.
It displays a bunch of errors of one or more of these:
    Updating geometry: Error build geometry(11): Geom_TrimmedCurve::U1 == U2
    Invalid solution from DogLeg solver.
    Updating geometry: Error build geometry(11): Geom_TrimmedCurve::U1 == U2
    Invalid solution from LevenbergMarquardt solver.
    Updating geometry: Error build geometry(15): Geom_TrimmedCurve::U1 == U2
    Invalid solution from SQP(augmented system) solver.
    Updating geometry: Error build geometry(12): Geom_TrimmedCurve::U1 == U2
    Invalid solution from BFGS solver.
It turns the entire sketch red.  If it does this, pressing undo *twice* (it stays red if you only hit it once) then trimming the previous two segments in the opposite order may work.  The double undo issue is likely a separate bug.
A C++ exception error:
    Unhandled unknown C++ exception in ViewProvider::eventCallback (Event type: SoLocation2Event, object type: SketcherGui::ViewProviderSketch)
    (this may be an unrelated bug)
It works as expected.  This is by far the least common outcome.  Unfortunately, this holds true for my experience with freecad in general.

Most relevant bit of trace:
$ ./FreeCAD_weekly-builds-42474-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.1.0, Libs: 1.1.0devR42474 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

QColorSpace attempted constructed from invalid primaries: QPointF(0,0) QPointF(0,0) QPointF(0,0) QPointF(0,0)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bfd0) [0x7fc36de6dfd0]
#1  0x7fc305487ca3 in Part::GeomCurve::intersect(Part::GeomCurve const*, std::vector<std::pair<Base::Vector3<double>, Base::Vector3<double> >, std::allocator<std::pair<Base::Vector3<double>, Base::Vector3<double> > > >&, double) const from /tmp/.mount_FreeCAbGNFFM/usr/lib/./Part.so+0x23
#2  0x7fc2cfd03160 in Sketcher::SketchObject::deriveConstraintsForPieces(int, std::vector<int, std::allocator<int> > const&, std::vector<Part::Geometry const*, std::allocator<Part::Geometry const*> > const&, Sketcher::Constraint const*, std::vector<Sketcher::Constraint*, std::allocator<Sketcher::Constraint*> >&) const from /tmp/.mount_FreeCAbGNFFM/usr/lib/Sketcher.so+0x3b0
#3  0x7fc2cfd033b1 in Sketcher::SketchObject::deriveConstraintsForPieces(int, std::vector<int, std::allocator<int> > const&, Sketcher::Constraint const*, std::vector<Sketcher::Constraint*, std::allocator<Sketcher::Constraint*> >&) const from /tmp/.mount_FreeCAbGNFFM/usr/lib/Sketcher.so+0x151
#4  0x7fc2cfd03721 in createNewConstraintsForTrim(Sketcher::SketchObject const*, int, std::array<int, 2ul> const&, std::array<Base::Vector3<double>, 2ul> const&, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> >&, std::vector<Sketcher::Constraint*, std::allocator<Sketcher::Constraint*> >&, std::set<int, std::greater<void>, std::allocator<int> >&) from /tmp/.mount_FreeCAbGNFFM/usr/lib/Sketcher.so+0x241
#5  0x7fc2cfd1a1b7 in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /tmp/.mount_FreeCAbGNFFM/usr/lib/Sketcher.so+0x527
#6  0x7fc2cfdebc21 in Sketcher::SketchObjectPy::trim(_object*) from /tmp/.mount_FreeCAbGNFFM/usr/lib/Sketcher.so+0x91
#7  0x7fc2cfdebe74 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /tmp/.mount_FreeCAbGNFFM/usr/lib/Sketcher.so+0x24

xtemp09, 2025-07-06:
I can say that both reports seem to be the same case. It looks like the commit 263331f6fb6c80416bc0d5f3560b6a39c9e50485 (the PR #15811) has to do with this issue. Sadly, the description to the commit says: 

> DO NOT MERGE because not complete: not all constraints are not transferred.
> 
> Includes:
> 
> * Adjust `deriveConstraintsForPieces` for trim
> * Make sure new points are constrained on the cutting edges
> * Add constraints at trim end(s) dependent on pre-exisiting ones with cutting curve(s). Also change tests to reflect that auto-coincidence can now be applied to non-periodic curves as well.
> 
> Known issue(s):
> * Some redundant constraints may be added under certain circumstances (e.g. in
> presence of perpendicular-via-point)

The last line seems to be the exact cause of the crash. The commentary says the same:

https://github.com/FreeCAD/FreeCAD/blob/db11f481e9ac74c8f39c513a5cc233058af20fb1/src/Mod/Sketcher/App/SketchObject.cpp#L3326-L3329

luzpaz, 2025-07-06:
cc @AjinkyaDahale

lagnat, 2025-07-06:
It's worth pointing out that @AjinkyaDahale said the `DO NOT MERGE` comment was incorrect and that he was going to edit that out.  It feels premature to bring out the pitchforks.

AjinkyaDahale, 2025-07-06:
@luzpaz thanks for letting me know.

@xtemp09 thanks for finding out the exact place where this happens. I'll get to solving it at some point, but I predict it's one of the "spend 20 hours to find one line to change" kind of issue.

@floofywolf I tend to keep adding `TODO`s as I code; they shouldn't be taken as a sign that this shouldn't be merged. `FIXME`s this applies less so, but at some point I decided to ask to have it merged. Trim was becoming a big headache for me and there was little chance I could figure out everything without having it merged and used by the community.

I should repeat that trim before this was an 800+ line behemoth and already had plenty of open issues. The solution I came up with was to essentially rewrite the whole thing, including adding some reusability with split etc. This has even solved some of the issues in the older trim, but added a bunch of corner cases, which I'm squashing as I'm made aware of them.

@lagnat thanks for reporting and the test case. I do recall saying the `DO NOT MERGE` should be removed but not where, and evidently it had more corner cases.

xtemp09, 2025-07-06:
Several days ago, I could not compile _FreeCAD_. Using `git bisect`, I identified the specific commit where the problem first appeared. Interestingly, this commit itself doesn't cause the bricking; rather, it simply uncovers the underlying issue. This situation seems to be similar.

xtemp09, 2025-07-07:
> And another trim bug

@floofywolf, if you find another bug, it is worth creating another issue. In this case it won't be lost.

luzpaz, 2025-07-17:
@floofywolf please create a new ticket, TIA!

PR #22577 (2025-07-24): Sketcher: Avoid crash when using trim as described in #22352

## Issues

Fixes #22352

## Before and After Images

N/A
commit 018ac7b9 (2025-07-24): Sketcher: Skip checks in `deriveConstraintsForPieces` when risky

Possibly fixes #22352.

It is possible to call the method when new geos are not yet created or added to
the sketch. In such cases, the values obtained can be "dirty", and is likely a
seg-fault.