FreeCAD/FreeCAD #22353: TechDraw: FillTemplateFields tool is broken

FEA-eng, 2025-07-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The [TechDraw FillTemplateFields](https://wiki.freecad.org/TechDraw_FillTemplateFields) tool is broken in the latest Windows weekly. The following errors appear when enabling it:

```
18:34:16  Running the Python command 'TechDraw_FillTemplateFields' failed:
Traceback (most recent call last):
  File "D:\FreeCAD_weekly-builds-42474-conda-Windows-x86_64-py311\Mod\TechDraw\TechDrawTools\CommandFillTemplateFields.py", line 59, in Activated
    TechDrawTools.TaskFillTemplateFields()
  File "D:\FreeCAD_weekly-builds-42474-conda-Windows-x86_64-py311\Mod\TechDraw\TechDrawTools\TaskFillTemplateFields.py", line 192, in __init__
    self.cb3.setChecked(QtCore.Qt.Checked)

'PySide6.QtWidgets.QAbstractButton.setChecked' called with wrong argument types:
  PySide6.QtWidgets.QAbstractButton.setChecked(CheckState)
Supported signatures:
  PySide6.QtWidgets.QAbstractButton.setChecked(arg__1: bool, /)
```

Safe mode doesn't help. It works in 1.0.1.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42474 (Git)
Build date: 2025/07/04 04:10:37
Build type: Release
Branch: main
Hash: 927d302401caad6a9e518584e0665ae9f61509f2
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-07-05:
I try my best, see https://forum.freecad.org/viewtopic.php?p=835164#p833501

Syres916, 2025-07-05:
@luzpaz just to clarify, the code was never designed/tested on Qt6 so it's not just gone into a regression state this last month, it's never worked on Qt6.x, there's only two users who have tried it since the Weeklies converted from Qt5.15.x to Qt6.x!!!

FEA-eng, 2025-07-05:
> I try my best, see https://forum.freecad.org/viewtopic.php?p=835164#p833501

Thanks, I haven't noticed that forum discussion. It works when I replace TaskFillTemplateFields.py with your file. Can you submit a PR to fix it?

Syres916, 2025-07-05:
> Can you submit a PR to fix it?

Yep, I'll sort it this evening.

Syres916, 2025-07-05:
Created two PRs one for `main` #22355 and one for backport #22356 and tomorrow I'll have a look for any other `isChecked` lurking in the code.

PR #22355 (2025-07-06): [TechDraw] Fix Qt6 compatability of FillTemplateFields tool
`QCheckBox` `setChecked` => `setCheckState` , I'll review the rest of the code and if found, raise PR(s) as required.

Fixes #22353
closed 2025-07-06 by kadet1090