FreeCAD/FreeCAD #22358: Rotate/Polar Transform behaves unexpected

svenakela, 2025-07-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using the Rotate/Polar Transform function in Sketcher, the distribution gets inverted when a degree value is specified. The arrayed object is supposed to be evenly distributed over 315 degress but as you can see in the video clip it doesn't.

The function has behaved like this a long time, it's not only the latest version.

https://github.com/user-attachments/assets/e6826c2d-c6ab-41d8-b2de-bc8d7e027a7a

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.42474 (Git) AppImage
Build date: 2025/07/04 04:10:37
Build type: Release
Branch: main
Hash: 927d302401caad6a9e518584e0665ae9f61509f2
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/163.331/1
```

### Subproject(s) affected?

None

### Anything else?

It would be very pleasant if the function automatically distributes the objects over a full circle if the value is set to 360, including the copied object to avoid overlaying duplicates.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-07-06:
Apparently the "breaking" angle is 200 degrees. 

[image]

Version:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * Curves 0.6.64
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * sheetmetal 0.7.22
  * Silk 0.2.0

maxwxyz, 2025-07-06:
@PaddleStroke

PaddleStroke, 2025-07-07:
I can confirm on my dev build.

It was not happening in 1.0

This is likely a regression due to the recent changes to OVP.
@tetektoza