FreeCAD/FreeCAD #22360: problem with FEM solver

Roberto8303, 2025-07-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

why can't I perform the calculation?

### Full version info

```shell
OS: Windows 10 build 19042
Architecture: x86_64
Version: 1.1.0dev.42474 (Git)
Build date: 2025/07/04 04:10:37
Build type: Release
Branch: main
Hash: 927d302401caad6a9e518584e0665ae9f61509f2
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Object center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/113.872/1.5
Installed mods: 
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * PieMenu 1.12.0
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-06:
Can you describe what the issue is and post a demo file.
Could you try if it happens with the builds from: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Roberto8303, 2025-07-06:
problem with decimals with comma or dot, when running solver with Calculix

14:59:04  camassets.INFO: CAM asset directory changed in preferences: User parameter:BaseApp/Preferences/Mod/CAM/Tools ToolPath C:\Users\rob\AppData\Roaming\FreeCAD\CamAssets
14:59:05  camassets.INFO: CAM asset directory changed in preferences: User parameter:BaseApp/Preferences/Mod/CAM/Tools ToolPath C:\Users\rob\AppData\Roaming\FreeCAD\CamAssets
14:59:06  camassets.INFO: CAM asset directory changed in preferences: User parameter:BaseApp/Preferences/Mod/CAM/Tools ToolPath C:\Users\rob\AppData\Roaming\FreeCAD\CamAssets
14:59:07  
14:59:07  Check prerequisites...
14:59:07  
14:59:07  Get mesh data for constraints, materials and element geometry...
14:59:07  [{'ccx_elset': 'Evolumes', 'ccx_elset_name': 'MaterialSolidSolid', 'mat_obj_name': 'MaterialSolid', 'ccx_mat_name': 'Steel-C40E'}]
14:59:07  ConstraintFixed:
14:59:07      Type: Fem::ConstraintFixed, Name: ConstraintFixed
14:59:07      ReferenceShape ... Type: Face, Object name: Pad, Object label: Pad, Element name: Face5
14:59:07  ConstraintForce:
14:59:07      Type: Fem::ConstraintForce, Name: ConstraintForce
14:59:07      ReferenceShape ... Type: Face, Object name: Pad001, Object label: Pad001, Element name: Face11
14:59:07  Getting mesh data time: 0.203 seconds.
14:59:07  
14:59:07  CalculiX solver input writing...
14:59:07  Input file:C:/Users/rob/Desktop/prova flessione fem\SolverCcxTools\FEMMeshNetgen.inp
14:59:07  One monster input file.
14:59:07  Writing time CalculiX input file: 0.031 seconds.
14:59:07  run CalculiX at: C:\Users\rob\Downloads\FreeCAD_weekly-builds-42474-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-42474-conda-Windows-x86_64-py311\bin/ccx.exe with: C:/Users/rob/Desktop/prova flessione fem\SolverCcxTools\FEMMeshNetgen.inp
14:59:08  CalculiX done without error!
14:59:08  
14:59:08  CalculiX read results...
14:59:08  Read ccx results from frd file: C:/Users/rob/Desktop/prova flessione fem\SolverCcxTools\FEMMeshNetgen.frd
14:59:08  loading results failed
14:59:08  could not convert string to float: ' 1,23085E+02'

FreeCAD_weekly-builds-42474-conda-Windows-x86_64-py311

maxwxyz, 2025-07-06:
can you provide the example file?
@FEA-eng FYI

Roberto8303, 2025-07-06:
where can i upload it?

maxwxyz, 2025-07-06:
You can drag and drop the .zip of your FreeCAD file in a new comment. If it is too large you might need an external file hosting service and link it here.

FEA-eng, 2025-07-06:
It's a known issue. Try changing the FreeCAD language to English, switching the decimal separator to dot in Windows settings or using CalculiX binary from [here](https://www.dhondt.de/calculix_2.22_4win.zip) instead of the one provided with FreeCAD.

maxwxyz, 2025-07-06:
@FEA-eng is it this: https://github.com/FreeCAD/FreeCAD/issues/10865 or which issue? Maybe we can update it.

Roberto8303, 2025-07-06:
I tried to set the language to English but it always gives an error

maxwxyz, 2025-07-06:
I think you need to change number format in FreeCAD preferences or system locale.

FEA-eng, 2025-07-06:
> [@FEA-eng](https://github.com/FEA-eng) is it this: [#10865](https://github.com/FreeCAD/FreeCAD/issues/10865) or which issue? Maybe we can update it.

More like #18368. For some reason, the "could not convert string to float" error became very common recently. I also experienced it. I just used CalculiX binary from the original website to avoid it. I think that FreeCAD should also provide that binary instead of the current one due to issues like thius or #10865.

> I did it but it doesn't change, does the file sent to you work?

Only with non-default ccx binary.

Roberto8303, 2025-07-06:
How do I change the usage of the external binary file you linked me?

FEA-eng, 2025-07-06:
> How do I change the usage of the external binary file you linked me?

Edit --> Preferences --> FEM --> CalculiX --> CalculiX binary (uncheck "Search in known binary directories and select the downloaded and unpacked ccx_static.exe file as ccx binary path).

closed 2025-07-06 by maxwxyz