FreeCAD/FreeCAD #22391: Crash with tapered pad

TechnischeEntwicklungenGretsch, 2025-07-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Create Sketch of circle with 2mm or smaller diameter, make pad of any length while adding negative Taper angle. As soon as confirming the taper angle, freecad crashes. Console spits out the following:

[NoCrash.txt](https://github.com/user-attachments/files/21122951/NoCrash.txt)

Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3def0) [0x7fbdde64def0]
#1  0x7fbda5ac75b8 in MAT_ListOfBisector::Current() const from /usr/lib/libTKTopAlgo.so.7.9+0x8
#2  0x7fbda5ae2af5 in MAT2d_Mat2d::CreateMat(MAT2d_Tool2d&) from /usr/lib/libTKTopAlgo.so.7.9+0x3025
#3  0x7fbda5b0042c in BRepMAT2d_BisectingLocus::Compute(BRepMAT2d_Explorer&, int, MAT_Side, GeomAbs_JoinType, bool) from /usr/lib/libTKTopAlgo.so.7.9+0x9ec
#4  0x7fbda2c60b61 in BRepFill_OffsetWire::Init(TopoDS_Face const&, GeomAbs_JoinType, bool) from /usr/lib/libTKBool.so.7.9+0x1d1
#5  0x7fbda2c6105e in BRepFill_OffsetWire::BRepFill_OffsetWire(TopoDS_Face const&, GeomAbs_JoinType, bool) from /usr/lib/libTKBool.so.7.9+0xfe
#6  /usr/lib/libTKOffset.so.7.9(+0x69c95) [0x7fbda4d1dc95]
#7  0x7fbda4d1f5db in BRepOffsetAPI_MakeOffset::Perform(double, double) from /usr/lib/libTKOffset.so.7.9+0x6fb
#8  0x7fbda62f76f5 in Part::ExtrusionHelper::createTaperedPrismOffset(TopoDS_Wire, gp_Vec const&, double, bool, TopoDS_Wire&) from /usr/lib/freecad/lib/Part.so+0x1e5
#9  0x7fbda62f9fac in Part::ExtrusionHelper::makeElementDraft(Part::ExtrusionParameters const&, Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> >&, Base::Reference<App::StringHasher>) from /usr/lib/freecad/lib/Part.so+0x3bc
#10  0x7fbd9d510c5a in PartDesign::FeatureExtrude::buildExtrusion(Base::Flags<PartDesign::FeatureExtrude::ExtrudeOption>) from /usr/lib/freecad/lib/_PartDesign.so+0x20ca
#11  0x7fbde131a722 in App::DocumentObject::recompute() from /usr/lib/freecad/lib/libFreeCADApp.so+0xc2
#12  0x7fbda61088f3 in Part::Feature::recompute() from /usr/lib/freecad/lib/Part.so+0x13
#13  0x7fbde12959b4 in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x3f4
#14  0x7fbde129c12a in App::Document::recomputeFeature(App::DocumentObject*, bool) from /usr/lib/freecad/lib/libFreeCADApp.so+0x11a
#15  0x7fbd96ba0743 in PartDesignGui::TaskExtrudeParameters::tryRecomputeFeature() from /usr/lib/freecad/lib/PartDesignGui.so+0x13
#16  /usr/lib/libQt6Core.so.6(+0x1d37ef) [0x7fbddedd37ef]
#17  0x7fbde252159c in Gui::QuantitySpinBox::valueChanged(double) from /usr/lib/freecad/lib/libFreeCADGui.so+0x4c
#18  0x7fbde2522fce in Gui::QuantitySpinBox::updateFromCache(bool, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0xce
#19  0x7fbde2523160 in Gui::QuantitySpinBox::updateText(Base::Quantity const&) from /usr/lib/freecad/lib/libFreeCADGui.so+0x70
#20  0x7fbde25234b3 in Gui::QuantitySpinBox::stepBy(int) from /usr/lib/freecad/lib/libFreeCADGui.so+0x83
#21  0x7fbde012d4a1 in QAbstractSpinBox::wheelEvent(QWheelEvent*) from /usr/lib/libQt6Widgets.so.6+0x91
#22  0x7fbddff5ae84 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x544
#23  0x7fbddff01c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x90
#24  0x7fbddff062ea in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x16aa
#25  0x7fbde1ec3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#26  0x7fbdded68118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#27  /usr/lib/libQt6Widgets.so.6(+0x178b5d) [0x7fbddff78b5d]
#28  0x7fbddff01c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x90
#29  0x7fbde1ec3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#30  0x7fbdded68118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#31  0x7fbddf59c3df in QGuiApplicationPrivate::processWheelEvent(QWindowSystemInterfacePrivate::WheelEvent*) from /usr/lib/libQt6Gui.so.6+0x15f
#32  0x7fbddf6248f4 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x2f4
#33  /usr/lib/libQt6XcbQpa.so.6(+0x5a0c7) [0x7fbdd8f660c7]
#34  /usr/lib/libglib-2.0.so.0(+0x5e87d) [0x7fbddd3c787d]
#35  /usr/lib/libglib-2.0.so.0(+0x5fcd7) [0x7fbddd3c8cd7]
#36  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7fbddd3c8ee5]
#37  0x7fbddf03c5e2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x72
#38  0x7fbdded744b6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x206
#39  0x7fbdded6c7c1 in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0xb1
#40  0x7fbde1dc7bf5 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x9b5
#41  freecad(+0x8276) [0x560ff1603276]
#42  /usr/lib/libc.so.6(+0x276b5) [0x7fbdde6376b5]
#43  /usr/lib/libc.so.6(__libc_start_main+0x89) [0x7fbdde637769]
#44  freecad(+0x8625) [0x560ff1603625]

(https://github.com/user-attachments/files/21122962/NoCrash.FCMacro.txt)

### Full version info

```shell
OS: Garuda Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, OCC 7.9.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * InventorLoader 1.5.1
  * sheetmetal 0.7.22

Issue persists in safe mode
```

### Subproject(s) affected?

None

### Anything else?

Any other diameter above 2mm (i tested 2.1mm, 3mm, 10mm, 50mm) work just fine giving the expected result of a cone, as can be seen with macro file. As soon as the diameter of the circle in the sketch is set to 2mm and sketch is exited, FreeCAD crashes.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-08:
@TechnischeEntwicklungenGretsch any chance you can test on 1.1-dev[^1] ? 

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

luzpaz, 2025-08-29:
@pendelf have you encountered this crash at all ?

luzpaz, 2026-08-30:
@robertely what verison of OCC will this be fixed in ?

stefano2734, 2026-09-12:
> [@robertely](https://github.com/robertely) what verison of OCC will this be fixed in ?

OCCT 8.0.2 in October 2026

luzpaz, 2026-09-12:
@pieterhijma per https://github.com/FreeCAD/FreeCAD/issues/22391#issuecomment-5644674360 this can be considered completed in the OCCT Liason project, right ?

stefano2734, 2026-09-13:
With solved in 8.0.2 available in october 2026, 

From progress  perhaps to waiting in this liaison board.

ElementW, 2026-09-13:
Can be considered a duplicate of #9708.
I tested, both that dupe and this issue are solved as of OCC 8.1.0.dev1-3d097a0328 (IR branch, FC 13b07d70e6e27ee61f8894531c22aeeb46aa93ed).

closed 2026-09-16 by luzpaz