FreeCAD/FreeCAD #22394: Workbenches in alphabetical order

G0rocks, 2025-07-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hey there :)
I'm wondering if we could have the workbenches in the dropdown be in alphabetical order since it always takes me 2 extra seconds to find the workbench I'm looking for.
Thanks :)

Here's how they currently look for me:
[image]

Also, just wanted to say, I love this program, thank you very much.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.7
  * Manipulator 1.5.7
  * Render 2024.1.23
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

oursland, 2025-07-08:
You can choose the order of the Workbenches in preferences:

`Preferences...`->`Workbenches`->`Available Workbenches` and drag them up or down to change their order.

G0rocks, 2025-07-08:
🤦 
Thank you very much, I did not know that.
Thanks!

closed 2025-07-08 by G0rocks