FreeCAD/FreeCAD #22421: Pixi build fails

jimcc, 2025-07-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Pixi build fails in Debian 12 on my uptodate FreeCAD build box.  The last build was prior to Commit 9c91b95.  Pixi has always been successfully built until now. This is my first issue report and I am a novice at all of this but this seems to be a real issue. The error that crops up is always the same:
Submodule path 'src/3rdParty/GSL': checked out '2828399820ef4928cc89b65605dca5dc68efca6e'
Submodule path 'src/3rdParty/OndselSolver': checked out '09d6175a2ba69e7016fcecc4f384946a2f84f92d'
Submodule path 'src/Mod/AddonManager': checked out '34d433a02c7ec5c73bec9c57d0a27ea70b36c90d'
Submodule path 'tests/lib': checked out 'f8d7d77c06936315286eb55f8de22cd23c188571'
Error: × expected string or table, found array
╭─[/home/jimcchristiansen/FreeCAD-Build-Home/FreeCAD-July-09-2025/FreeCAD/pixi.toml:178:9]
177 │ configure = [{ task = "configure-debug" }]
178 │ build = [{ task = "build-debug" }]
· ──────────────────────────
179 │ install = [{ task = "install-debug" }]
╰────

Error: × expected string or table, found array
╭─[/home/jimcchristiansen/FreeCAD-Build-Home/FreeCAD-July-09-2025/FreeCAD/pixi.toml:178:9]
177 │ configure = [{ task = "configure-debug" }]
178 │ build = [{ task = "build-debug" }]
· ──────────────────────────
179 │ install = [{ task = "install-debug" }]
╰────

./Compile-Test-FreeCAD-build-time-test-time.sh: 18: cd: can't cd to /home/jimcchristiansen/FreeCAD-Build-Home/FreeCAD-July-09-2025/FreeCAD/build/bin/
./Compile-Test-FreeCAD-build-time-test-time.sh: 25: ./FreeCAD: Permission denied

This is the script I use. I'm in Crostini right now but this occurs exactly the same on my other x86 Debian build box.
I'm adding my little script (renamed to .txt) that I use to build FC.

[Compile-Test-FreeCAD-build-time-test-time.txt](https://github.com/user-attachments/files/21165185/Compile-Test-FreeCAD-build-time-test-time.txt)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Generic/xwayland)
Architecture: x86_64
Version: 1.1.0dev.42475 (Git)
Build date: 2025/07/04 19:24:09
Build type: Unknown
Branch: main
Hash: f72397ce193b607e0555badc6c200c0bd7e61220
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/96.237/1.25
Installed mods: 
  * freecad.gears 1.3.0
  * sheetmetal 0.7.23
  * Curves 0.6.63
  * fasteners 0.5.38
  * SaveAndRestore 0.2.1
  * Pyramids-and-Polyhedrons
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jimcc, 2025-07-10:
The Full version info is copied from my last successful build 42475

Syres916, 2025-07-10:
@jimcc for info I raised it on the Pull Request for the change see https://github.com/FreeCAD/FreeCAD/pull/22174#issuecomment-3057548365

JohnOCFII, 2025-07-11:
Have you tried deleted your build directory and then re-running `pixi run configure` and `pixi run build?`

jimcc, 2025-07-11:
> Have you tried deleted your build directory and then re-running `pixi run configure` and `pixi run build?`

Hi John.  This is my script I use to build FreeCAD a couple times a week.  It has worked for months. I'm including it just to show what caused the responses.  I haven't bothered to learn how to parse the current date into the commands.  I just edit the file each time before running it. I know it's inelegant but it works for me :-)

#!/bin/sh
#XDG_SESSION_TYPE=X11
sudo systemctl restart networking.service
git clone --recurse-submodules   https://github.com/FreeCAD/FreeCAD.git
cd FreeCAD
XDG_SESSION_TYPE=xwayland
export QT_QPA_PLATFORM=xcb
pixi run configure
cd ..
date > build-time.txt
cd FreeCAD
pixi run build -j8
cd ..
date >> build-time.txt
XDG_SESSION_TYPE=xwayland
export QT_QPA_PLATFORM=xcb
#date > test-time.txt
cd /home/jimcchristiansen/FreeCAD-Build-Home/FreeCAD-July-09-2025/FreeCAD/build/bin/
#date > test-time.txt
#cd build/bin/
XDG_SESSION_TYPE=xwayland
export QT_QPA_PLATFORM=xcb
export KMP_AFFINITY=disabled
date > /home/jimcchristiansen/FreeCAD-Build-Home/FreeCAD-July-09-2025/test-time.txt
./FreeCAD -t0
#pixi run freecad -t0
date >> /home/jimcchristiansen/FreeCAD-Build-Home/FreeCAD-July-09-2025/test-time.txt

zbynekwinkler, 2025-07-11:
What is your version of pixi? It is just a hunch but

> Error: × expected string or table, found array
╭─[/home/jimcchristiansen/FreeCAD-Build-Home/FreeCAD-July-09-2025/FreeCAD/pixi.toml:178:9]
177 │ configure = [{ task = "configure-debug" }]
178 │ build = [{ task = "build-debug" }]
· ──────────────────────────
179 │ install = [{ task = "install-debug" }]
╰────

maybe it expects string or table because it is old? The CI uses

```
pixi-version: v0.48.1
```

closed 2025-07-11 by jimcc

jimcc, 2025-07-11:
jimcchristiansen@penguin:~/FreeCAD-Build-Home/FreeCAD-July-11-2025/FreeCAD/build$ pixi -V
pixi 0.49.0

zbynekwinkler, 2025-07-11:
> Should I file an issue with FreeCAD -t0 tests crashing

In general - yes. 

Just in this case YMMV since the crash is not in freecad code. But maybe there can be a workaround of some kind.