FreeCAD/FreeCAD #22423: TechDraw: Rasterization warning when zooming too much

pierreporte, 2025-07-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When you zoom on a line in TD as much as possible, up to making the whole screen black because the thickness is then larger than the screen, a lot of Qt warning about rasterization appears, identical to the following except for the values in `QRectF()`.

    (qt.svg.draw) Shape of type 12 ignored because it will take too long to rasterize (bounding rect= QRectF(-3.96609e+06,-2.99903e+06 9.23371e+06x6.5955e+06) ). Set QT_SVG_DISABLE_SIZE_LIMIT=1 to disable this check.

Zooming that much is useless in practice but a warning is not needed IMO. FreeCAD should either refuse to zoom further or not display the warning at all. In any case, that level of zoom does not allow to see anything, but users may think it’s an error.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/07/07 16:45:59
Build type: Release
Branch: main
Hash: d2a4f8994fba9e3e62098f1ff9a1af67e0ecf6e1
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * Curves 0.6.63
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.22
  * FusedFilamentDesign 0.25.220
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

fflipski, 2025-07-10:
https://github.com/user-attachments/assets/31ece5d1-a4af-44bc-a3a7-74b374934cd5

no issue here on Mint on latest Dev... but I am still using QT5.  Haven't figure out how to get QT6 compile yet.

OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42477 (Git)
Build date: 2025/07/05 20:44:22
Build type: Unknown
Branch: main
Hash: db11f481e9ac74c8f39c513a5cc233058af20fb1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.4.20250315, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.9.0
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/93.2987/1

WandererFan, 2025-07-11:
That message string does not occur in the TD source.   Looks like a message from Qt.

luzpaz, 2025-09-27:
@fflipski any progress with Qt6 ?

fflipski, 2025-09-28:
> [@fflipski](https://github.com/fflipski) any progress with Qt6 ?

Nope. but I am busy with work right now. Should have time to look at it in the next few weeks,