FreeCAD/FreeCAD #22429: gesture functionality not the same as touchpad

ChrisCompulife, 2025-07-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

# Bug Report: Assembly Dragging Disabled in Gesture Navigation Style

## Summary
Assembly components cannot be dragged when using Gesture Navigation style, while the same dragging operations work correctly with Touchpad Navigation style.

## Environment
- **FreeCAD Version**: OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42339 (Git)
Build date: 2025/06/26 08:11:26
Build type: Release
Branch: main
Hash: 352fadd5c794e6560fc8259005a3276507f0c33a
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/109.57/1
Installed mods: 
  * FreeGrid 2.2.0
  * freecad.gears 1.3.0
  * btl 0.9.9
  * Curves 0.6.63
  * AIWorkbench
  * fasteners 0.5.38
  * OpenTheme 2025.5.20
  * CurvedShapes 1.0.13

- **Navigation Style**: Gesture Navigation (problematic) vs Touchpad Navigation (working)
- **Workbench**: Assembly workbenches (A2plus, Assembly3, Assembly4, etc.)

## Description
When working with assemblies in FreeCAD, users expect to be able to drag and manipulate assembly components directly in the 3D viewport. This functionality works correctly when using Touchpad Navigation style but fails when using Gesture Navigation style.

## Steps to Reproduce
1. Open FreeCAD and create or load an assembly with moveable components
2. Set navigation style to "Gesture Navigation" (Edit → Preferences → Display → 3D View → Navigation)
3. Attempt to drag an assembly component using left-click and drag
4. **Observe**: Component does not move/drag as expected
5. Change navigation style to "Touchpad Navigation"
6. Attempt the same drag operation
7. **Observe**: Component moves correctly

## Expected Behavior
Assembly components should be draggable regardless of the navigation style selected, maintaining consistent user interaction across all navigation modes.

## Actual Behavior
Assembly dragging is non-functional in Gesture Navigation style, forcing users to switch navigation styles to perform assembly operations.

## Root Cause Analysis
The issue appears to stem from conflicting mouse input handling between two systems:

1. **Gesture Navigation Input Handling**: Gesture navigation intercepts mouse movements to detect and interpret gestures, particularly left-click-drag operations for gesture recognition.

2. **Assembly Manipulation Input Handling**: Assembly workbenches also require left-click-drag operations to move and manipulate components.

The gesture recognition system appears to have higher priority in the input handling chain, consuming mouse events before they can reach the assembly manipulation code.

## Technical Details
- **Mouse Button Mapping Conflict**: Both gesture recognition and assembly dragging compete for the same mouse input (left-click-drag)
- **Event Handling Priority**: Gesture navigation likely processes mouse events earlier in the event chain
- **Navigation Style Architecture**: Different navigation styles use different mouse button mappings and event handling approaches

## Impact
- **Workflow Disruption**: Users must manually switch navigation styles when working with assemblies
- **User Experience Inconsistency**: Different navigation styles provide different functional capabilities
- **Productivity Loss**: Additional manual steps required for assembly work

## Suggested Solutions

### Short-term Workarounds
1. Allow assembly dragging with alternative mouse button combinations in Gesture Navigation (e.g., Ctrl+left-drag, middle-click-drag)
2. Provide gesture sensitivity controls that can be temporarily disabled during assembly operations

### Long-term Solutions
1. **Input Event Prioritization**: Implement context-aware input handling that prioritizes assembly operations when in assembly workbenches
2. **Navigation Style Unification**: Ensure all navigation styles support equivalent functionality for core operations like assembly manipulation
3. **Configurable Mouse Mappings**: Allow users to customize mouse button assignments per navigation style to avoid conflicts

### Implementation Approach
- Modify the input event handling chain to check for assembly context before processing gestures
- Add configuration options for gesture sensitivity or temporary gesture disable
- Implement alternative input methods for assembly dragging in gesture navigation mode

## Priority
**Medium-High** - This affects core functionality and forces users into suboptimal workflows.

## Related Components
- Navigation system (gesture recognition)
- Assembly workbenches (A2plus, Assembly3, Assembly4)
- 3D viewport input handling
- User preferences system

## Additional Notes
This issue demonstrates the need for better coordination between navigation systems and workbench-specific interactions. The solution should maintain the benefits of gesture navigation while preserving full assembly functionality.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42339 (Git)
Build date: 2025/06/26 08:11:26
Build type: Release
Branch: main
Hash: 352fadd5c794e6560fc8259005a3276507f0c33a
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/109.57/1
Installed mods: 
  * FreeGrid 2.2.0
  * freecad.gears 1.3.0
  * btl 0.9.9
  * Curves 0.6.63
  * AIWorkbench
  * fasteners 0.5.38
  * OpenTheme 2025.5.20
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

iplayfast, 2025-07-11:
Yeah, "maintaining consistent user interaction" is  not correct. It should be "maintaining consistent user functionality"

iplayfast, 2025-07-12:
Can we just add Ctrl+lmb so that it passes through to the underlying workbench. That is if the gesture (or whatever) doesn't have anything assigned as ctrl+select it ignores it?

iplayfast, 2025-11-24:
I'm on linux. Others are apparently able to make it work, I'm not. [https://forum.freecad.org/viewtopic.php?p=858891#p858891](url)

luzpaz, 2025-11-29:
> Basically same as https://github.com/FreeCAD/FreeCAD/issues/14139. In my opinion workbenches should be able to add states and transitions to the navigation style and also disable transitions or reorder priority. This way a workbench specific feature can get priority. It is however a bit of a refactor for which I still don't have time.

Close as a duplicate or label as one ?

sciunto, 2025-12-02:
The difference i see with Roy_043 on the forum is that he is using wayland and I'm using X11.

sciunto, 2026-01-01:
> I'm on mint.

Check this, it might help you: https://github.com/FreeCAD/FreeCAD/issues/14139#issuecomment-3703406510

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

closed 2026-10-31 by maxwxyz