FreeCAD/FreeCAD #22433: Weekly build for MacOS fails

julian7, 2025-07-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The last two weekly builds weren't finished for osx-arm64. Example: https://github.com/FreeCAD/FreeCAD/actions/runs/15981181616/job/45082620302

```
1 identity imported.
keychain: "/Users/runner/work/_temp/app-signing.keychain-db"
version: 256
class: 0x00000010 
...
Error: HTTP status code: 403. A required agreement is missing or has expired. This request requires an in-effect agreement that has not been signed or has expired. Ensure your team has signed the necessary legal agreements and that they are not expired.

This process stores your credentials securely in the Keychain. You reference these credentials later using a profile name.

Validating your credentials...
Error: Process completed with exit code 1.
```

osx-64 is throwing the same error now, but up until build 23, it threw a different error.

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-07-12:
@oursland? ☝️

oursland, 2025-07-13:
@chennes @yorikvanhavre Any update on the renewed developer license?

julian7, 2025-07-19:
Apparently, Jul 14 weekly build has MacOS arm64 image, this issue is most likely fixed. There is still a build issue on the osx-64 build, but that's unrelated.

oursland, 2025-07-20:
Yes, this specific issue is resolved.  The other issue remains.

@maxwxyz

chennes, 2025-07-20:
(I'm leaving a note here for future reference -- the error wasn't actually that the certificate had expired, but that @yorikvanhavre and I *both* needed to log into our Apple dev accounts and accept their new terms of service -- once we had both done that, the problem resolved itself)