FreeCAD/FreeCAD #22435: no dimension text or value while adding a new dimension in TechDraw

carloolicad, 2025-07-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I cannot visualize the dimensions value when I create a new dimensions in Techdraw workbenck, being it linear or diameters or angles. I had not this problem one week ago. Sometimes rebooting solves the problem, sometimes not.
In the attached zipped FCStd file "disco", you can see dimension are inserted without value. You can select them from the Tree but not from the Page001. When I print the page I get some black square where the value of the dimension is supposed to be (see pdf attached)

[disco.pdf](https://github.com/user-attachments/files/21179969/disco.pdf)

. It affect old files, where I can see old dimensions value but not those of the new ones I try to insert.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.18
  * dodo 1.0.1
  * dxf-library
  * ExplodedAssembly
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Help 1.0.3
  * POV-Ray-Rendering
  * Render 2024.12.15
  * sheetmetal 0.7.22
  * slic3r-tools
```

### Subproject(s) affected?

TechDraw

### Anything else?

[disco.zip](https://github.com/user-attachments/files/21180002/disco.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-07-11:
I am not able to reproduce the issue.

When opening the sample file, the font size of dimension is set to 200mm, after changing it to reasonable 10mm it looks like this:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

luzpaz, 2025-07-12:
> probably the "font" type and "fontsize" for the new dimensions changed.

@carloolicad what font and fontsize did they change to? It would be good to know this.