FreeCAD/FreeCAD #22444: `File->Merge Document`  shows  `<ElementMap> ElementMap.cpp(486): Missing element name prefix id` warnings

ErtugrulEge, 2025-07-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyone!
Steps to reproduce:
1.Creaate a simple pad with part design and save the file
2.Close all files 
3.`File->Merge Document` and select the file you saved and press open
I have 3 same warning.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42531 (Git)
Build date: 2025/07/11 16:00:50
Build type: Release
Branch: main
Hash: 759bad7455a3a554cdf48a8182709cad991c8fe5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-13:
@drwho495 toponaming related?

luzpaz, 2025-07-13:
@ErtugrulEge any chance you could provide the example file for convenient confirmation by testers/dev ?

ErtugrulEge, 2025-07-13:
I added a sample file to issue report

luzpaz, 2025-07-13:
@semhustej would you mind testing this BIM related ticket ? TIA

drwho495, 2025-07-13:
> @drwho495 toponaming related?

Yeah probably, I can look at it in a few hours

ErtugrulEge, 2025-08-30:
I am still getting this warning message I also tried with Part design and Draft 
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
```

[image]

semhustej, 2025-08-30:
I can reproduce the issue too.

Note: User has to have `Report view` open to see the warning messages
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * manifest.json
  * Reporting
  * Telemetry 1.0.2

```

ErtugrulEge, 2026-10-15:
I can't reproduce. Seems fixed 

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 597.06/NVIDIA Corporation/Quadro RTX 4000/PCIe/SSE2
Installed mods:
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.81
```

closed 2026-10-15 by ErtugrulEge