FreeCAD/FreeCAD #22448: TechDraw: Broken view missing piece of source shape

WandererFan, 2025-07-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In a broken view of a complex App::Part, a portion of an included sub-App::Part is missing.   This problem was originally reported here: https://forum.freecad.org/viewtopic.php?t=98265

[image]

[BrokenView_missingPieceOfSource.FCStd.not.zip](https://github.com/user-attachments/files/21196369/BrokenView_missingPieceOfSource.FCStd.not.zip)

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42349 +18 (Git)
Build date: 2025/07/11 22:29:23
Build type: Unknown
Branch: ComplexSectionErrors2
Hash: 74beb1017525b2df5073c438cebfd6ed99c4e830
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, boost 1_74, Eigen3 3.4.0, PySide 5.15.2
shiboken 5.15.2, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.9.0
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 115/141.585/1
Installed mods: 
  * fasteners 0.5.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-07-17:
@FreeCAD/design-working-group - opinions on broken view of shells?

pierreporte, 2025-07-17:
I think everything should be included in every kind of view. Of course there could be a parameter.

WandererFan, 2025-08-11:
This should have been closed by PR #22633.

closed 2025-08-11 by WandererFan