FreeCAD/FreeCAD #22457: FC 1.0.1 'typing_extensions' error on loading BIM

tbuegel, 2025-07-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Fresh installation and setup basic configuration then starting BIM throws ModuleNotFoundError: No module named 'typing_extensions'

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

Full text in report view:
22:14:10  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\BIM\nativeifc\ifc_observer.py", line 85, in slotChangedDocument
    from nativeifc import ifc_tools  # lazy import
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\Mod\BIM\nativeifc\ifc_tools.py", line 36, in <module>
    from ifcopenshell import geom
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'typing_extensions'

There was nothing else done, only the clicking on BIM/Architecture icon after fresh installation of FreeCAD 1.0.1
```

### Subproject(s) affected?

None

### Anything else?

I'm new to FreeCAD, so I don't know whether the operations of the toolt inspite of the error are operating normal or not. But that stopped me from familiarizing myself with FreeCAD with a video tutorial.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-07-12:
Duplicate of #21422

semhustej, 2025-07-13:
Thanks for reporting. I think this many users are encountering this issue.

closed 2025-07-13 by Roy-043