FreeCAD/FreeCAD #22458: Regression: relative font path doesn't work

luzpaz, 2025-07-12:
Unfortunately it does not work:

[image]
left: FreeCad, right: bash showing project structure
 <br />
 <br />

**Version**

OS: Ubuntu 25.04 (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.0.0.
Build type: Release
Python 3.13.2, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ExplodedAssembly
  * FreeGrid 2.2.0
  * FusedFilamentDesign 0.25.220
  * Manipulator 1.6.0
  * STEMFIE 0.3.1
  * SvgWorkbench 1.0.0.dev6
  * fasteners 0.5.38
  * freecad.gears 1.3.0

_Originally posted by @rubienr in https://github.com/FreeCAD/FreeCAD/issues/17819#issuecomment-3065998895_

* Old ticket: #5746
* PR that fixed it #17819

luzpaz, 2025-07-12:
@rubienr can you also test on 1.1-dev ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

Roy-043, 2025-07-13:
I have checked with a  similar tree (with similar directory names) and I can't confirm. The issue may be due to file rights, or a problem with the TTF.

rubienr, 2025-07-13:
Works with  [FreeCAD_weekly-builds-42531-conda-Linux-x86_64-py311.AppImage](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-42531-conda-Linux-x86_64-py311.AppImage) from weekly build [0b40df3](https://github.com/FreeCAD/FreeCAD-Bundle/commit/0b40df3032e1218cdf6dc2cc47d3c5301735232f).

[image]

Roy-043, 2025-07-13:
@rubienr This feature was indeed introduced in V1.1.

Let's Close this then.

closed 2025-07-13 by Roy-043

rubienr, 2025-07-13:
Font file permissions are okay.

Tested with Ubuntu latest (does not work):

[image]

Error: 
```python
Traceback (most recent call last):
  File "/usr/share/freecad/Mod/Draft/draftobjects/shapestring.py", line 139, in execute
    char = Part.makeWireString("L", obj.FontFile, 1, 0)[0]
<class 'Part.OCCError'>: Can not open font file: ./hauora-sans_5.2.5/ttf/hauora-sans-latin-800-normal.ttf
ShapeString00.: Can not open font file: ./hauora-sans_5.2.5/ttf/hauora-sans-latin-800-normal.ttf
```

Font file:

```bash
tree
.
├── hauora-sans_5.2.5
│   ├── LICENSE
│   └── ttf
│       ├── hauora-sans-latin-700-normal.ttf
│       └── hauora-sans-latin-800-normal.ttf
└── rummy.FCStd
```

Permission:
```bash
ls -n ./hauora-sans_5.2.5/ttf/hauora-sans-latin-800-normal.ttf
-rw-rw-r-- 1 1000 1000 136640 Mar  2 18:38 ./hauora-sans_5.2.5/ttf/hauora-sans-latin-800-normal.ttf
```

Roy-043, 2025-07-13:
This is a V1.1 feature. You can therefore not expect it to work in an earlier version.

https://wiki.freecad.org/Release_notes_1.1#Draft_Workbench
https://wiki.freecad.org/Draft_ShapeString#Relative_font_path