FreeCAD/FreeCAD #22470: It seems export a wrong dxf file.

HeibaiPanda, 2025-07-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

this is the draw I want to export

[image]

the dxf file is  like this

[image]

the centerlines seem offset

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-15:
Could you test if this happens in the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
@furgo16 related to current DXF changes?

HeibaiPanda, 2025-07-15:
I exported a dxf file again using the weekly version.
It looks the same.

[image]

[image]

closed 2025-10-26 by chennes

semhustej, 2025-11-05:
The OP issue is fixed in the latest weekly, thank you.

I still get issues with some of the exported dimensions though. 

[image]

Exported file (remove the zip extension):

[test-export-fixed.dxf.zip](https://github.com/user-attachments/files/23373811/test-export-fixed.dxf.zip)

Should I open a new issue?

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251104 (Git shallow)
Build date: 2025/11/04 19:07:44
Build type: Release
Branch: main
Hash: af014a1bb44264d749ba9522917f0d1ac76c4bbf
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```