FreeCAD/FreeCAD #22471: Issue with Coincident Constraint on Circle in Sketcher

stemyn12, 2025-07-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I create a circle in the Sketcher and then try to draw a line connected to it, the Coincident constraint appears to work at first. However, when I zoom in, I can see that the line is actually slightly offset and not truly connected to the circle.
As a result, I’m unable to create a valid face from the sketch.
The Sketch validation tool also detects this issue.

I’ve recorded a video demonstrating the problem.

https://youtu.be/UjdYpX1OcZA

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * boltsfc 2022.11.5
  * fasteners 0.5.38
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://youtu.be/UjdYpX1OcZA

[Issue with Coincident Constraint on Circle in Sketcher.FCStd.zip](https://github.com/user-attachments/files/21206572/Issue.with.Coincident.Constraint.on.Circle.in.Sketcher.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-07-13:
try decreasing convergence in the advanced sketch panel

ErtugrulEge, 2025-07-14:
Sorry for interruption  but what is "advanced sketch panel" ?

Roy-043, 2025-07-14:
The valid face problem is probably due to T-connections:
https://wiki.freecad.org/Sketcher_Workbench#Profile_sketches

stemyn12, 2025-07-14:
> This is a tessellation issue. The circle is represented as a polygon on the screen, but internally the exact circle is used.
> 
> You can change the "Segments per geometry" preference for a smoother display. https://wiki.freecad.org/Sketcher_Preferences#Display
> 

I changed the setting from 100 to 1000, but the problem still persists.

[image]

closed 2025-07-14 by maxwxyz

Roy-043, 2025-07-15:
@stemyn12 Please start a forum topic if you need more help.