FreeCAD/FreeCAD #22474: Regarding the inability to drill screw holes

yukimaru7777, 2025-07-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

[image]

[image]

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yukimaru7777, 2025-07-14:
Thank you for your reply maxwxyz, here is the file in question.

[回転サーボと曲げるサーボの接続素材.zip](https://github.com/user-attachments/files/21212803/default.zip)

luzpaz, 2025-07-14:
@yukimaru7777 zip was not attached.  
Can you also test on FreeCAD version 1.1-dev?  (as max pointed out above)

yukimaru7777, 2025-07-14:
Thank you for pointing that out. I realized I had only edited my previous comment and didn’t actually attach the ZIP file. My apologies for the confusion.

Here is the ZIP file containing the model in question:
[回転サーボと曲げるサーボの接続素材.zip](https://github.com/user-attachments/files/21217684/default.zip)

I’ll also test the issue with the 1.1-dev version as suggested and follow up with the results here.

Thanks again for your support!

FEA-eng, 2026-01-15:
Recompute fails in this file and the sketch has invalid reference:

```
09:36:59  <App> PropertyTopoShape.cpp(507): Recomputation required for document 'test' on geo element version change in test#Body.Shape: 1.4 -> 0.15.70200.5
09:37:00  <PropertyLinks> PropertyLinks.cpp(518): test#Sketch002.AttachmentSupport missing element reference test#Hole ;#40:2;:H11fb,F.Face2
09:37:00  PositionBySupport: AttachEngine3D: subshape not found Hole.?Face2
09:37:00  Sketch002: AttachEngine3D: subshape not found Hole.?Face2
```

But when I reattach it, the pocket only makes 4 holes on the sides of the cuboid. There's no such rectangular cutout in the middle.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260113 (Git shallow)
Build date: 2026/01/13 22:48:50
Build type: Release
Branch: (HEAD detached at 4cc0a1b)
Hash: 4cc0a1b718735be10261d936066cf1382dd51884
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```