FreeCAD/FreeCAD #22475: placement of Constraint

develmissions, 2025-07-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello, I drew the following figure using Python. Now I want to move the created constraints so that they are not on the figure and have the right position. Please guide me. 

[image]

### Full version info

```shell
I couldn't find any section or documentation where I could use Python code to change the size from the original shape.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

entuland, 2026-08-27:
setLabelPosition() and setLabelDistance() on the Sketch object are able to alter such values

https://github.com/FreeCAD/FreeCAD/blob/e13677fd60ba1fe801516e36b38fba67ccc09f9c/src/Mod/Sketcher/App/SketchObject.pyi#L587

https://github.com/FreeCAD/FreeCAD/blob/e13677fd60ba1fe801516e36b38fba67ccc09f9c/src/Mod/Sketcher/App/SketchObject.pyi#L614

luzpaz, 2026-08-30:
@entuland can you show a PoC ?

entuland, 2026-08-30:
@luzpaz sorry not sure what a Proof of Concept would be here, I was just pointing out that the operation can be already performed from python.

Given a sketch with a point and a constraint, running the setLabelDistance() on the sketch object would affect the label distance as if you did that manually (you need to perform some interaction with the sketch to have it update when running it in the console, but that's irrelevant, as scripting the whole thing as with the OP case, that would be done by the document recompute).

```python
doc = App.ActiveDocument
sketch = doc.getObject("Sketch")
sketch.getLabelDistance(0) #just to print the old value
sketch.setLabelDistance(0, 20)
sketch.getLabelDistance(0)
```

https://github.com/user-attachments/assets/fc1e7dc6-6bfe-4407-90f5-cb2bdc112140