FreeCAD/FreeCAD #22480: WebGL html file - Running offline

kileereet, 2025-07-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I need to generate a fully offline WebGL html file.
Currently the file loads the threejs scripts from online.
Please add a option to integrate the threejs scripts directly inside the html. So it can run fully offline.

### Full version info

```shell
FreeCad 1.0.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-07-14:
Related to https://github.com/FreeCAD/FreeCAD/pull/21695

maxwxyz, 2025-07-14:
Does #21695 solve the issue?

kileereet, 2025-08-07:
Yes template can help to create an offline working html by bundling all the threejs things inside the template.