FreeCAD/FreeCAD #22501: MacOS 13 Weekly builds failing

matthiasdanner, 2025-07-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The macOS 13 builds are failing, for example https://github.com/FreeCAD/FreeCAD/actions/runs/16275274645/job/45960661841#step:9:1429

> $SRC_DIR/src/Base/XMLTools.cpp:139:18: error: aligned allocation function of type 'void *(std::size_t, std::align_val_t, const std::nothrow_t &) noexcept' 
>  │ │ is only available on macOS 10.13 or newer
>  │ │   139 |     auto ptr = ::operator new(static_cast<size_t>(size),
>  │ │       |                  ^

the reason is, the aligned constructor doesn't exist before macOS 10.13

it seems to be configured correctly here: https://github.com/FreeCAD/FreeCAD/blob/7e03fc662097b1a07d3dc9a84f488e82a46fedfe/package/rattler-build/osx/launcher/CMakeLists.txt#L4

but on the weekly build there is somehow the version set to 10.9 (https://github.com/FreeCAD/FreeCAD/actions/runs/16275274645/job/45960661841#step:9:489)

> -DCMAKE_OSX_DEPLOYMENT_TARGET=10.9

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

matthiasdanner, 2025-07-15:
in the pixi.lock file there is a osx >= 10.9 dependency. maybe that's the problem. can't say for sure because I don't know pixi

https://github.com/FreeCAD/FreeCAD/blob/7e03fc662097b1a07d3dc9a84f488e82a46fedfe/pixi.lock#L2236

chennes, 2025-07-15:
@oursland any idea?

matthiasdanner, 2025-07-22:
Something changed. The last weekly now failed for a different reason. And Windows and macOS arm64 builds failed as well

oursland, 2025-07-22:
Indeed!  I am able to reproduce locally on macOS arm64 and am looking into the underlying cause.

luzpaz, 2025-07-25:
Can this be closed ?

oursland, 2025-07-25:
I would like to double-check that the next build succeeds.  My previous PR that was supposed to fix the issue did not.

oursland, 2025-07-26:
@luzpaz #22741 did fix the problem and the issue may now be closed.

closed 2025-07-26 by luzpaz