FreeCAD/FreeCAD #22510: Std TransformManip is disabled in Arch SectionPlane tool

markkomartin, 2025-07-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

All BIM and DRAFT objects can be modified using the Std TransformManip tool. 
Std TransformManip tool is disabled for the ARCH section plane object and for dimension tool. 

In the weekly build version 1.1.0dev.41694 (Git)(2025/05/13) Std TransformManip tool was deactivated.
In the weekly build version 1.1.0dev.42133 (Git) (2025/06/09), the Std TransformManip tool was activated but was then immediately deactivated. 
In the weekly build version 1.1.0dev.42523 (Git) (2025/07/07)Std TransformManip tool was deactivated.

The affected commits should be: 
https://github.com/FreeCAD/FreeCAD/pull/21342
https://github.com/FreeCAD/FreeCAD/pull/21878

Std TransformManip tool is not very useful for the dimension tool. 
Std TransformManip tool, on the other hand, is very useful for aligning an existing section plane with any inclined plane. 
Currently, it's possible to align the section to any plane by previously setting a working plane.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42523 (Git)
Build date: 2025/07/07 16:45:59
Build type: Release
Branch: main
Hash: d2a4f8994fba9e3e62098f1ff9a1af67e0ecf6e1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1
```

### Subproject(s) affected?

BIM

### Anything else?

[image]

[image]

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

markkomartin, 2025-07-16:
> > In the weekly build version 1.1.0dev.41694 (Git)(2025/05/13) Std TransformManip tool was deactivated.
> 
> The referenced PR was merged on May 26, 13 days after the tool was reportedly deactivated => [#21342 (comment)](https://github.com/FreeCAD/FreeCAD/pull/21342#issuecomment-2908938137)
> 
> It cannot have been the cause, then. I'll have a look regardless.

Thanks @furgo16!!
Sorry for my poor English. 
On May 13th, the Std TransformManip tool was unavailable for Arch's SectionPlane tool.
I noticed that after your commit ( May 26 ) , you were able to enable it, but it had issues with movement.
Tetektoza then fixed positioning and transparency issues of Arch section tool , and Std TransformManip tool became disabled again.
I have reported the two commits so maybe it is easier to understand how you managed to activate Std TransformManip tool.
You're doing a great job with BIM workbench. 
Thanks!

furgo16, 2025-07-16:
> In the weekly build version 1.1.0dev.41694 (Git)(2025/05/13) Std TransformManip tool was deactivated.

Question: were these two context menu entries _ever_ enabled? E.g. before the build version mentioned above.

furgo16, 2025-07-16:
Thanks for checking! I can't promise I'll manage, but I'll look into enabling them.

Roy-043, 2026-01-29:
This indeed has to do with the base objects some BIM and Draft objects are based on. I am going to close this one in favor of #27138.