FreeCAD/FreeCAD #22515: inconsistent units conversion of Units.schemaTranslate()

dprojects, 2025-07-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Inconsistent units conversion of Units.schemaTranslate()

```
import FreeCAD
from FreeCAD import Units
pq = FreeCAD.Units.parseQuantity
```

now if you convert "500 mm" to Building units you have string:

```
>>> Units.schemaTranslate(Units.Quantity("500 mm"), 5)[0]
'1\' 7" + 5/8"'
```

but this string syntax is incorrect so you can't convert it back:

```
>>> pq('1\' 7" + 5/8"')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ValueError: invalid unit expression
```

the string should be **'1\' (7 + 5/8)"'**:

```
>>> pq('1\' (7 + 5/8)"')
498.475 mm
```

### Full version info

```shell
[code]
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Woodworking 1.0.1.39285.1.1.1.2025.07.15
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-16:
> Version: 0.21.2.33771 (Git) AppImage 

@dprojects ??

Roy-043, 2025-07-16:
This is still an issue in the current dev version.
https://github.com/FreeCAD/FreeCAD/issues/11345#issuecomment-3015256645

dprojects, 2025-07-16:
Yes, this bug is still there, you can test it from python console like this:

```
>>> import FreeCAD
>>> from FreeCAD import Units
>>> t = Units.schemaTranslate(Units.Quantity("500 mm"), 5)[0]
>>> t
'1\' 7" + 5/8"'
>>> pq(t)
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ValueError: invalid unit expression: '1' 7" + 5/8"'

>>> t = '1\' (7 + 5/8)"'
>>> pq(t)
498.475 mm
>>> 
```

```
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git) AppImage
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/100.968/1
Installed mods: 
  * Woodworking 1.0.1.39285.1.1.1.2025.07.16

```

dprojects, 2025-07-16:
I [fixed this directly in the library](https://github.com/dprojects/Woodworking/commit/add900581107efa6f4cc74d8f724f8e981e58b73), so no user action is needed and also this is no longer bug from my perspective.

closed 2025-07-16 by dprojects