FreeCAD/FreeCAD #22517: Core: Origin objects label should stay 'Origin'

PaddleStroke, 2025-07-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

I think it is not helpful that the origin objects have unique Labels. I don't see the point for it. What's your thoughts on this?

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-07-16:
I agree completely. Basically everything is named globally. The scope is not taken into account.

pierreporte, 2025-07-16:
For the strange numbering, I think it would be best to add numbers from the first object but not as heavy as 001. In Catia it's Object.1, then Object.2, etc.

pieterhijma, 2025-07-17:
> I agree completely. Basically everything is named globally. The scope is not taken into account.

It is actually the case that everything is named globally. The tree view shows a hierarchy but in actuality there is no such hierarchy. For example, consider these two different ways of creating a group:

```python
doc = App.newDocument("hierarchy")
doc.addObject("App::DocumentObjectGroup", "Group")
box = doc.addObject("Part::Box", "Box")
doc.Group.addObject(box)
```

The view provider of a DocumentObjectGroup apparently "claims" its child "Box" which results in Box1 becoming a child in the tree view.  In actuality you can still reference Box1 with `doc.Box`. Note that `Box` is listed in `doc.Group.Group`.

In this more crude way of creating a group, there is no such view provider and as such, the document object is not claimed and reflects the relations without a hierarchy:

```python
doc = App.newDocument("nohierarchy")
doc.addObject("App::FeaturePython", "Group")
doc.Group.addExtension("App::GroupExtensionPython")
box = doc.addObject("Part::Box", "Box")
doc.Group.addObject(box)
```

Note that also in this case Box is part of `doc.Group.Group`, however, the Box is not shown as part of the hierarchy of Group. 

My point is that the "scope" or hierarchy is simply an illusion of the Gui and in the App side of things, there is no such hierarchy. All document objects are available in the scope of the document object.