FreeCAD/FreeCAD #22521: Sketch move of blocked elements

bjnobre, 2025-07-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Look at this annoying bug: constrained locked lines keeps moving when trimming.

https://github.com/user-attachments/assets/e52f88e1-3854-4354-a091-4fbb5acbe956

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.42325 (Git) AppImage
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR) [ OS: Portuguese/Brazil (pt_BR) ]
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * Glass
  * sheetmetal 0.7.24
  * fasteners 0.5.38
  * IconThemes
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-17:
Does it work in 1.0.1? Can you please provide an example file?
@AjinkyaDahale FYI

Roy-043, 2025-07-17:
AFAIK this is intended behavior.
[The block constraint only affects the freely movable parts of an edge. Blocked edges can have other constraints, and applying additional constraints to a blocked edge can modify it.](https://wiki.freecad.org/Sketcher_ConstrainBlock)

maxwxyz, 2025-07-18:
Agreed, but trimming them should not move the points of the edges IMO.

chennes, 2025-07-24:
I can't reproduce -- although the trim does cause the blocked constraint itself to be lost I think that's the same behavior as older versions. I'm attaching the file I tested with, trying to replicate @bjnobre 's demonstration. When I go through the sequence shown in the video, I don't get anything moving, just trimming as expected. This was on  
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42669 (Git)
Build date: 2025/07/24 02:08:02
Build type: Release
Branch: replaceTraceback
Hash: d49b94f0ca1940813e8ff46aed13cb2e7d8dbeb8
Python 3.13.5, Qt 6.9.1, Coin 4.0.4, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.8.1
```

[PrepToTrim.zip](https://github.com/user-attachments/files/21398741/PrepToTrim.zip)

maxwxyz, 2025-07-29:
I cant reproduce either. Closing then.

```
OS: macOS 13.7.6
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/07/28 17:17:34
Build type: Release
Branch: main
Hash: 8dc6397fc4433785f73f8a1958ff11243c348376
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/72/1
Installed mods: 
  * freecad.gears 1.3.0
  * CfdOF 1.31.4
  * Defeaturing 1.2.6
  * Gridfinity 0.11.9
  * AddonManager 2025.7.17
  * Manipulator 1.6.0
  * fasteners 0.5.39
  * lattice2 1.0.0
  * fcVM.backup1752306371.4383721 (Disabled)
  * Telemetry 1.0.2
  * CurvedShapes 1.0.13
  * Cables 0.2.1
  * FusedFilamentDesign 0.25.250
  * Curves 0.6.64

```