FreeCAD/FreeCAD #22534: Return height for G81,82,83 broken by commenting out.

J-Dunn, 2025-07-17:
https://github.com/FreeCAD/FreeCAD/blame/8acbb224fc7d7fef32eabbebb28459477e920be6/src/Mod/CAM/Path/Post/scripts/grbl_post.py#L600

Why was this done?

closed 2025-07-17 by J-Dunn

Syres916, 2025-07-17:
I think it was by accident due to a pre-commit invalid rule. If you create a PR to reverse that specific section of code you may well see in the CI Tests if it alerts to the syntax etc if it's not happy. Such a big PR where most of it was automated is always a risk IMHO.

J-Dunn, 2025-07-17:
Yeah , I realise this is to try to enforce some kind of coding standard for the project and this needs to be automated. 

I think it has trouble following some of the PY-C++ interface magic. 

Also sometimes it split fns across several lines but if I split it, it wants to compact it to one line. 
sometimes if enforces two line breaks above a fn, which I like to do myself. Yet the other it removed one.  It was probably progammed by a female ;)