FreeCAD/FreeCAD #22535: Core: Expressions using Label is a blocker for non-unique Labels.

PaddleStroke, 2025-07-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently you can create expressions using either the Name property or the Label property. So if you have a Body that you rename 'Stick' you can do : 

`Body.Length`  Or `<<Stick>>.Length`

**Why is it a problem?**
If you are using non-unique Labels then the expression doesn't know what object to use.

**Why would you want non-unique Labels?**
To fix https://github.com/FreeCAD/FreeCAD/issues/12139
and https://github.com/FreeCAD/FreeCAD/issues/22517

**Additional issue :** 
For the unsuspecting user it is kind of confusing to use the property editor. Because the autocomplete offers him multiple possibility. And if he has not read the wiki and know freecad very well, he doesn't know there are both a Name (hidden from the user anyway) and a Label. So take this example : 

[image]

When you know it make sense. When you don't it's confusing.

**So what can we do? Here is a suggestion, please let me know what you think: **

The object Name is hidden from the user. So what is the point of having the capability of referencing the Name?

IN THE EDITOR :
- We remove the possibility to reference by Name in the Expression Editor.
- We remove the <<>> from referencing by Label.
- If more than one object have the same Label, we either add the path ("Body001.Origin") or an indentation number if they have the same path ("Body <1>").

So, in my previous example the user sees only : 

[image]

IN THE BACKEND :
We always store the expression using the NAME of the object. Not the Label. So when the editor is closed, we replace the user-facing string by one that uses the Name of the object. And the other way around, when the editor is opened we replace the Name based string by the friendly Label based one.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mnesarco, 2025-07-17:
I can say the opposite: non-unique Labels is a blocker for Expressions. 
As FreeCAD supports identifying objects by labels, labels should be unique, any identifier must be unique.

PaddleStroke, 2025-07-17:
@3xxx we could keep the `<<>>` syntax available as a way to force storing the label and not the Name. This way that usecase is saved.

mnesarco, 2025-07-17:
I have a different question: What is the use case for non-unique labels? what problem does it solves?

kaiwas, 2025-07-17:
Removing the existing functionality is a bad suggestion.

Maybe it could look like this?

[image]

The drop-down list always displays "internal names", as well as names specified by the user, even if it does not match the entered text.

That is, searching by internal name has priority, because this name is always there.

mnesarco, 2025-07-17:
Naming is hard. As this thing probably get its way to main anyway, please ensure that it is backwards compatible at all levels. There are thousands of hours already invested in models with expressions, macros and addons... ✌️

belinski-roman, 2025-07-17:
> Removing the existing functionality is a bad suggestion.
> 
> Maybe it could look like this?
> 
> [image]
> 
> The drop-down list always displays "internal names", as well as names specified by the user, even if it does not match the entered text.
> 
> That is, searching by internal name has priority, because this name is always there.
> 
> 
> At the same time, if the user searches by the specified name, the specified name will be at the top.
> 
> [image]
> 

Or like this[image]

belinski-roman, 2025-07-17:
But in general, I support you! If you switch to the system you proposed in the form of Body <1>, Body <2>, it would solve some of the problems.
For example, when using the Fasteners workbench, if you insert the same bolts, they appear as M6x25-Screw, M6x25-Screw001, and because of this, the BOM from the Assembly workbench considers them to be different objects and does not add them together (this is somehow handled in the Fasteners workbench, but that's a different issue).
Another problem that doesn't exist yet, but will inevitably arise as you develop a new tool for creating arrays. Assigning an index like
Body
|—Body <2>
|—Body <2>
|—Body <3>
would be useful if we wanted to later split the array into individual objects.

pieterhijma, 2025-07-17:
> Naming is hard. As this thing probably get its way to main anyway, please ensure that it is backwards compatible at all levels. There are thousands of hours already invested in models with expressions, macros and addons... ✌️

I second this: This would be a really big change with many consequences. Since label names are stored inside the document when references, we need a strategy for backward compatibility. In fact, I would argue that for such a change, we should first have a good way to do migrations.

belinski-roman, 2025-07-17:
After some thought, I came to the conclusion that this problem should be solved comprehensively!
It seems to me that it is worth leaving both unique labels and entering indexes of occurrences in the form <1>,<2>.
Moreover, I believe that we need another identifier for Configuration, for example (01), (02)! Well, or let's specify the letter Label instead of (01), (02).
This change affects too many parts of Freecad.:

1. Copy and paste an element. In this case, we most likely want to create a new part based on the copied one, and it is logical that a unique name is appropriate here.:

Body          <<Body>>
Body001    <<Body001>>
Body002    <<Body002>>

2. Occurrences of identical elements (App::Link or SubShapeBinder).
Here it is worth adding indexes <1>,<2> if there are more than one occurrences. It is also possible that Clone should be included here if the entire Body is specified as the Base Ftature, rather than an operation at a specific stage in the tree. The non-unique label is present:

Body<1>    <<Body>>
Body<2>    <<Body>>
Body<3>    <<Body>>

But there will probably be a dependency on the Copy On Change parameter, and if it is switched to Enable for all occurrences, then it should look like this again.:

Body          <<Body>>
Body001    <<Body001>>
Body002    <<Body002>>

The same applies to instances for an Array (show them as nested elements in a tree), but leave a non-unique label.:

Array Body    <<Array Body>>
|-Body<1>    <<Body>>
|-Body<2>    <<Body>>
|-Body<3>    <<Body>>

In the case when the array is destroyed, its elements should probably become references.

3. The occurrence of different variants of the same part
- App::Link or SubShapeBinder + Configuration Table
- Clone (if an operation in the tree is specified as the Base Ftature and the Clone differs from the original)
- Variant Parts, which is not available yet, but it is on the way
Indexes <1>,<2> and an identifier for Configuration (01), (02) are required:

Body<1>(00)    <<Body>>
Body<2>(01)    <<Body>>
Body<3>(02)    <<Body>>

Next, I assume that arrays will appear in the future, where each occurrence could be in the form of different Configurations (for example, in Solidworks)

Array Body          <<Array Body>>
|-Body<1>(00)    <<Body>>
|-Body<2>(01)    <<Body>>
|-Body<3>(02)    <<Body>>

Then let's say for BOM it would be possible to count occurrences forever, based on the index <1>,<2>, but if the occurrence has a different Configuration, then sum it up taking into account (01), (02).
The Fasteners workbench could be connected to this unified system, for example:

Screw DIN6912<1>(M6x25)
Screw DIN6912<2>(M6x25)
Screw DIN6912<1>(M6x60)
Screw DIN6912<1>(M8x25)
Screw DIN6912<2>(M8x25)
Where Screw DIN6912 is a non-unique name, <1> is the number of identical occurrences, and (M6x25) is Configuration.

This is a simple layman's view, I described how I see. I may have made some logical mistakes, but this is just an attempt to draw attention to the fact that the approach should be more thorough.

omnicortex, 2025-07-17:
👀

mnesarco, 2025-09-10:
Not suggesting `<<Name>>` in the autocomple will make Vars addon unusable.