FreeCAD/FreeCAD #22540: [FEM] FreeCAD crash when switching to FEM WB

muezabdalla, 2025-07-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

when I switch to FEM WB  freecad crash and give this error:
```
FreeCAD 1.1.0, Libs: 1.1.0devR42594 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

During initialization the error "No module named 'AddonCatalog'" occurred in /home/muez/applications/compiled/freecad/FreeCAD-git/build/Mod/AddonManager/./InitGui.py
Please look into the log file for further information
default matplotlib import
Unknown command 'Std_AddonMgr'
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x40110) [0x7f2324256110]
#1  /usr/lib/libpython3.13.so.1.0(PyDict_SetItem+0) [0x7f2324fb12d0]
#2  /usr/lib/libpython3.13.so.1.0(+0x23d24a) [0x7f232503d24a]
#3  /usr/lib/libpython3.13.so.1.0(+0x23db32) [0x7f232503db32]
#4  /usr/lib/libvtkWrappingPythonCore3.13-9.3.so.1(PyVTKTemplate_New+0x90) [0x7f2284652540]
#5  /usr/lib/python3.13/site-packages/vtkmodules/vtkCommonCore.so(+0x1be8b9) [0x7f226b1be8b9]
#6  /usr/lib/python3.13/site-packages/vtkmodules/vtkCommonCore.so(PyVTKAddFile_vtkArrayIteratorTemplate+0x16) [0x7f226b1bed66]
#7  /usr/lib/python3.13/site-packages/vtkmodules/vtkCommonCore.so(real_initvtkCommonCore+0x623) [0x7f226b35cee3]
#8  /usr/lib/libpython3.13.so.1.0(+0x288520) [0x7f2325088520]
#9  /usr/lib/libpython3.13.so.1.0(+0x286d30) [0x7f2325086d30]
#10  /usr/lib/libpython3.13.so.1.0(+0x287f74) [0x7f2325087f74]
#11  /usr/lib/libpython3.13.so.1.0(+0x1b8c33) [0x7f2324fb8c33]
#12  /usr/lib/libpython3.13.so.1.0(_PyEval_EvalFrameDefault+0x482d) [0x7f2324fe7a4d]
#13  /usr/lib/libpython3.13.so.1.0(+0x196795) [0x7f2324f96795]
#14  /usr/lib/libpython3.13.so.1.0(PyObject_CallMethodObjArgs+0xfc) [0x7f2324f976ac]
#15  /usr/lib/libpython3.13.so.1.0(PyImport_ImportModuleLevelObject+0x329) [0x7f2324ffae69]
#16  /usr/lib/libpython3.13.so.1.0(+0x1e0ed9) [0x7f2324fe0ed9]
#17  /usr/lib/libpython3.13.so.1.0(+0x1b89ef) [0x7f2324fb89ef]
#18  /usr/lib/libshiboken6.so.6.8(+0x35ec5) [0x7f2325f2cec5]
#19  /usr/lib/libpython3.13.so.1.0(+0x1b8b71) [0x7f2324fb8b71]
#20  /usr/lib/libpython3.13.so.1.0(_PyObject_MakeTpCall+0x6b) [0x7f2324f962ab]
#21  /usr/lib/libpython3.13.so.1.0(+0x118bdd) [0x7f2324f18bdd]
#22  /usr/lib/libpython3.13.so.1.0(_PyEval_EvalFrameDefault+0x71e1) [0x7f2324fea401]
#23  /usr/lib/libpython3.13.so.1.0(PyEval_EvalCode+0xae) [0x7f2325071dae]
#24  /usr/lib/libpython3.13.so.1.0(+0x270657) [0x7f2325070657]
#25  /usr/lib/libpython3.13.so.1.0(+0x1b89ef) [0x7f2324fb89ef]
#26  /usr/lib/libpython3.13.so.1.0(_PyEval_EvalFrameDefault+0x482d) [0x7f2324fe7a4d]
#27  /usr/lib/libpython3.13.so.1.0(+0x196795) [0x7f2324f96795]
#28  /usr/lib/libpython3.13.so.1.0(PyObject_CallMethodObjArgs+0xfc) [0x7f2324f976ac]
#29  /usr/lib/libpython3.13.so.1.0(PyImport_ImportModuleLevelObject+0x329) [0x7f2324ffae69]
#30  /usr/lib/libpython3.13.so.1.0(+0x1e0ed9) [0x7f2324fe0ed9]
#31  /usr/lib/libpython3.13.so.1.0(+0x1b89ef) [0x7f2324fb89ef]
#32  /usr/lib/libshiboken6.so.6.8(+0x35ec5) [0x7f2325f2cec5]
#33  /usr/lib/libpython3.13.so.1.0(+0x1b8b71) [0x7f2324fb8b71]
#34  /usr/lib/libpython3.13.so.1.0(_PyObject_MakeTpCall+0x6b) [0x7f2324f962ab]
#35  /usr/lib/libpython3.13.so.1.0(+0x118bdd) [0x7f2324f18bdd]
#36  /usr/lib/libpython3.13.so.1.0(_PyEval_EvalFrameDefault+0x71e1) [0x7f2324fea401]
#37  /usr/lib/libpython3.13.so.1.0(PyEval_EvalCode+0xae) [0x7f2325071dae]
#38  /usr/lib/libpython3.13.so.1.0(+0x270657) [0x7f2325070657]
#39  /usr/lib/libpython3.13.so.1.0(+0x1b89ef) [0x7f2324fb89ef]
#40  /usr/lib/libpython3.13.so.1.0(_PyEval_EvalFrameDefault+0x482d) [0x7f2324fe7a4d]
#41  /usr/lib/libpython3.13.so.1.0(+0x196795) [0x7f2324f96795]
#42  /usr/lib/libpython3.13.so.1.0(PyObject_CallMethodObjArgs+0xfc) [0x7f2324f976ac]
#43  /usr/lib/libpython3.13.so.1.0(PyImport_ImportModuleLevelObject+0x329) [0x7f2324ffae69]
#44  /usr/lib/libpython3.13.so.1.0(+0x1e0ed9) [0x7f2324fe0ed9]
#45  /usr/lib/libpython3.13.so.1.0(+0x1b89ef) [0x7f2324fb89ef]
#46  /usr/lib/libshiboken6.so.6.8(+0x35ec5) [0x7f2325f2cec5]
#47  /usr/lib/libpython3.13.so.1.0(+0x1b8b71) [0x7f2324fb8b71]
#48  /usr/lib/libpython3.13.so.1.0(+0x197441) [0x7f2324f97441]
#49  /usr/lib/libshiboken6.so.6.8(+0x3f2de) [0x7f2325f362de]
#50  /usr/lib/libpython3.13.so.1.0(+0x1b8b71) [0x7f2324fb8b71]
#51  /usr/lib/libpython3.13.so.1.0(_PyObject_MakeTpCall+0x6b) [0x7f2324f962ab]
#52  /usr/lib/libpython3.13.so.1.0(+0x118bdd) [0x7f2324f18bdd]
[... 90 lines cut ...]
Build type: Unknown
Branch: main
Hash: 3f6a3f10ffd617369fa9a12d23496f663c44a855
Python 3.13.5, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.1, boost 1_83, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.38
  * AnimationFreeCAD 1.0.0
  * FreeCAD_Assembly4.1 0.50.18
  * ExplodedAssembly
  * CfdOF 1.31.4
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

muezabdalla, 2025-07-20:
I tried to reproduce it on debian13 but I couldn't.

maxwxyz, 2025-07-21:
Can you reproduce this in safe mode? (Help->Restart in Safe Mode)

muezabdalla, 2025-07-21:
yes

luzpaz, 2025-07-21:
>  and it is not happening with the appimage version

Good clue. Can you add the AppImage [About](https://wiki.freecad.org/About) info for us to compare dependencies, @muezabdalla  ?

muezabdalla, 2025-07-21:
> Good clue. Can you add the AppImage [About](https://wiki.freecad.org/About) info for us to compare dependencies, [@muezabdalla](https://github.com/muezabdalla) ?
```
OS: Void Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git) AppImage
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.38
  * AnimationFreeCAD 1.0.0
  * FreeCAD_Assembly4.1 0.50.18
  * ExplodedAssembly
  * CfdOF 1.31.4
```

muezabdalla, 2025-07-27:
now when I try to replicate it I get this error:
```
FreeCAD is linked to a different VTK library than the imported VTK
python module. this is incompatible and will lead to errors.

wrong python module is imported from:
/user/lib/python3.13/site-packages/

No matching module was found in the current python path.
```
and it does not crash

the about info:
```
OS: Void Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.42736 (Git)
Build date: 2025/07/26 20:46:49
Build type: Unknown
Branch: main
Hash: 91641c4fb600381b645ed55e568e6c3d4f14f39f
Python 3.13.5, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.1, boost 1_83, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.38
  * AnimationFreeCAD 1.0.0
  * FreeCAD_Assembly4.1 0.50.18
  * ExplodedAssembly
  * CfdOF 1.31.4
```

luzpaz, 2025-07-27:
cc @FEA-eng any insight ?

FEA-eng, 2025-07-29:
> FreeCAD is linked to a different VTK library than the imported VTK
> python module. this is incompatible and will lead to errors.
> 
> wrong python module is imported from:
> /user/lib/python3.13/site-packages/
> 
> No matching module was found in the current python path.

@ickby Can you have a look at this?

muezabdalla, 2025-08-06:
I tried compiling freecad with `vtk-9.3.1` but compile it localy instead of using the system package for vtk. and that make freecad works fine. then I tried to use the same flags used when building the system package in void-linux and it gives:
`No module named vtkmodules`
and here is the flags:
```
-DBUILD_SHARED_LIBS=ON -DVTK_FORBID_DOWNLOADS=ON
 -DVTK_USE_EXTERNAL=ON
 -DVTK_MODULE_USE_EXTERNAL_VTK_utf8=OFF
 -DVTK_MODULE_USE_EXTERNAL_VTK_pegtl=OFF
 -DVTK_MODULE_USE_EXTERNAL_VTK_libharu=OFF
 -DVTK_MODULE_USE_EXTERNAL_VTK_exprtk=OFF
 -DVTK_MODULE_USE_EXTERNAL_VTK_ioss=OFF
 -DVTK_MODULE_USE_EXTERNAL_VTK_cgns=OFF
 -DVTK_MODULE_USE_EXTERNAL_VTK_verdict=OFF
 -DVTK_MODULE_USE_EXTERNAL_VTK_fast_float=OFF
 -DVTK_WRAP_PYTHON=ON -DVTK_PYTHON_VERSION=3
```

so in conclusion the problem is a packaging problem in the vtk package on void linux so I will close this issue

closed 2025-08-06 by muezabdalla