FreeCAD/FreeCAD #22541: BIM's WALL & SLAB do not calculate area when built from a SKETCH

jm98, 2025-07-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**As** an architect
**I want** to calculate the area/length both of walls and slabs created from complex profiles created via parameteric controlled constraints in sketcher
**So that** my model is parameteric controlled and I do not have to manually define coordinates when defining a polyline wire in BIM directly

At the moment this functionality is only available to wall/slab created from a BIM polyline.  The only partial workaround is to create a BIM polyline by tracing the sketch outline.  It took me 6 attempts to trace a 25 point outline and the BIM polyline is only a snapshot that does not respond to changes (manual and/or parametric) in the sketch.

To restate: BIM's "Wall" and "Slab" will create from a sketch but do not calculate length and area

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-07-18:
Would you mind submitting a file where the issue can be reproduced?

I could not reproduce the issue myself, neither on FreeCAD version 1.0.0 nor 1.1.0-dev. Here's the file I created for testing: [SlabArea_1-0.zip](https://github.com/user-attachments/files/21307194/SlabArea_1-0.zip). Both the slab with a sketch as base and that with a Draft wire as base show the same `HorizontalArea` and `VerticalArea` values from the Property Editor.

>  It took me 6 attempts to trace a 25 point outline and the BIM polyline is only a snapshot that does not respond to changes (manual and/or parametric) in the sketch.

You can do conversion from sketch to Draft object automatically by selecting the object and executing the https://wiki.freecad.org/Draft_Draft2Sketch, also available from the BIM workbench toolbar.

semhustej, 2025-07-18:
I am not able to reproduce the issue:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42531 (Git)
Build date: 2025/07/11 16:00:50
Build type: Release
Branch: main
Hash: 759bad7455a3a554cdf48a8182709cad991c8fe5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.0

```

jm98, 2025-07-18:
@Syres916 I was not aware of v1.0.1 :(

Please find attached

- zip file of model.  Issue is in the "Patio" group
- screengrab of property panel for slab "Slab From Wire"  (this is the BIM wire version)
- screengrab of property panel for slab "Slab from Sketch"  (Sketch "Patio Sketch")

[image]
[image]

[Slab Issue.zip](https://github.com/user-attachments/files/21313965/Slab.Issue.zip)

jm98, 2025-07-18:
Thank you @Syres916 

I'm still learning FreeCAD.  It looks like this is a PICNIC error (**P**roblem **I**n **C**hair **N**ot **I**n **C**omputer)

While creating the sketch to define the slab, it looks like that git some of the external geometry references referred to the blocks of the retaining wall (instead of the the wire defining the wall).

I've just performed a test and everything works as expected.  That is, if the retaining wall changes then the slab adjust automatically.

Thanks for all of the replies.

I'm closing this ticket

closed 2025-07-18 by jm98