FreeCAD/FreeCAD #22557: [Feature Request] passive installation parameter

Eibl-BAM, 2025-07-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It would be nice to have a passive installation parameter (/P maybe?). 
It's mostly used in enterprise environments to have an automated installation process like a remote deployment.
Having a silent parameter is great, but it doesn't give feed back to the end user if the FreeCAD is installed.

### Full version info

```shell
/
```

### Subproject(s) affected?

None

### Anything else?

*First time using Github :)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-09-27:
@Eibl-BAM is this akin to what a `--silent` flag would do ?

Eibl-BAM, 2025-09-30:
@luzpaz yes, the installation process would be the same. However, the installation displays a progress bar.

Eibl-BAM, 2025-11-06:
For more detail. The Windows installer has good installation parameter to use.

> https://learn.microsoft.com/en-gb/windows-server/administration/windows-commands/msiexec#display-options
 
For the /P parameter would be akin to the /qb parameter.