FreeCAD/FreeCAD #22558: Sketcher: When you enter a measurement value, different results are produced depending on whether you use a dot or a comma.

ErtugrulEge, 2025-07-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello everyone!
When you sketch, you can input dot(.) or comma(,) for values but they behave differently:

- if you enter (0.5)(zero point five ) it produces  5mm dimension ,

- if you enter (0,5)(zero comma five) it produces  0,5mm dimension

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-07-18:
That is to be expected as only one of the 2 is recognized as the decimal separator. This is standard throughout FreeCAD, with the Spreaadsheet WB being the only exception AFAIK.

There are several related preferences:
https://wiki.freecad.org/Preferences_Editor#General

kaiwas, 2025-07-18:
For example. Behavior in Blender.
When entering text fields (for example, search) - the enabled layout is accepted (for example, Russian).
But! When entering numeric fields - the Russian layout is automatically converted to English. As a result, there are no problems either when entering text, or when entering numbers, or even when using hot keys.
That is, in fact, it turns out that a specific language layout is used only when entering text in text fields.
This is very convenient. But users of one single layout probably do not understand the problem.

Video demonstration.

https://github.com/user-attachments/assets/91b74d13-c71c-4084-9dc2-aa6ee58bacb0

ErtugrulEge, 2025-07-18:
I have just searched this topic in general. Most of the world uses comma as seperator.I am on the comma region but I personally prefer dot. Even commercial programs has this problem. My suggestions:

*There can be a Decimal seperator symbol option (dot or comma) and then allow user to enter just one because allowing both cause can some performance issues.

*According to user settings FC shouldn't allow other option.For example, in comma region you should always enter comma, when you enter dot data shouldn't be processed.

Anyway, entering 5.2mm and getting 52mm is not a corect behaviour.

[image]

https://en.wikipedia.org/wiki/Decimal_separator
https://discourse.mcneel.com/t/display-dimensions-with-comma-instead-of-dot-2-54-instead-of-2-54/88951/2

kaiwas, 2025-07-18:
> some performance issues

it is unlikely that auto-replacement of one character can greatly affect performance.
However... performance is very bad in any case. Almost all operations are performed on one processor core. But that is another topic.

belinski-roman, 2025-07-19:
Look at the input fields in Inkscape

ErtugrulEge, 2025-07-19:
> Look at the input fields in Inkscape

Converts  dot to comma automatically. For example enter 75.5 Inkscape converts to 75,5

belinski-roman, 2025-07-19:
> > Look at the input fields in Inkscape
> 
> Converts  dot to comma automatically. For example enter 75.5 Inkscape converts to 75,5

In my opinion, this is the optimal solution. It seems to have the same behavior in Blender.

PaddleStroke, 2025-07-31:
I agree that the software should accept both , and . without failing like it is.

pierreporte, 2025-08-03:
I also think that thousands separators are pretty useless. There are more than just point and commas though (space and apostrophe), and in India they are placed differently (every 10 000). It’s a pain to be forced to type the comma when the program is set in a language that use the comma as a decimal separator. Ansys does this and it’s a huge pain. It’s better if the program can understand both the comma and point as decimal separator.

I’m discovering right know by reading https://github.com/FreeCAD/FreeCAD/issues/22558#issuecomment-3090400388 that Catia does like FreeCAD currently in the exact same situation but I’m always using the point because it’s easier with the numpad.

If we want to keep the separators, we could take inspiration from [speedcrunch](https://heldercorreia.bitbucket.io/speedcrunch/). It automatically detects the group separators from the decimal separator by just counting them (it even supports spaces and apostrophes, even mixed group separators), and defaults to the point as decimal separator when there is ambiguity. But it also displays the result as a tooltip so the user can see immediately if there is any mistake.

hyarion, 2025-08-03:
> So what the software does is that it strips the string from 'group separators'. So if your setting is 'decimal separator = .' then the , will be stripped. And you end up with the situation you describe.
> 
> So fixing this issue would mean drop support of group separators.

I agree with the issue, but I don't think we can just strip group separators. If someone pastes `10.000,01` or `10,000.01` then the program should be smart enough to understand that the user means _ten thousand and one hundredth_, just like `100,001` and `100.001` both means _one hundred and one thousandth_.

hyarion, 2025-08-03:
The smarter one try to make a system, the dumber it might end up...

The problem I'm trying to avoid is that people in countries that uses `,` officially, still might use `.` because of how different software behaves (like all programming languages).
If possible, I'd like the following behavior:

```py
for decimal_separator in [".", ","]:
    for thousands_separator in [".", ",", " "]:
        with (
            system_decimal_separator(decimal_separator),
            system_thousands_separator(thousands_separator),
        ):
            assertTrue(parse("10.01"), 10.01)
            assertTrue(parse("10,01"), 10.01)
            assertTrue(parse("10.000,01"), 10000.01)
            assertTrue(parse("10,000.01"), 10000.01)
            assertTrue(parse("10 000.01"), 10000.01)
```

MisterMakerNL, 2025-08-03:
We should also make a warning message when some American tries to switch to Imperial that it's stupid and they should feel bad,

flaviut, 2025-08-03:
FreeCAD itself shouldn't have special configuration or unique handling of this case; it should be something that the operating system takes care of for the user automatically. If the user wants to change the way these numbers are handled, they should change their settings in the system settings, and ideally then all programs should follow what they set.

Examples, and to be clear, the Python locale module is just a thin [wrapper around libc](https://www.gnu.org/software/libc/manual/html_node/Parsing-of-Integers.html) which is the library the operating system provides for this kind of functionality: 

`python -c 'import locale; locale.setlocale(locale.LC_ALL, ""); print(locale.format_string("%.3f", 1234567.89, grouping=True))'`:

| Locale | Formatted example |
|---|---|
|en_DK.UTF-8 | 1.234.567,890|
|en_US.UTF-8 | 1,234,567.890|
|fr_FR.UTF-8 | 1 234 567,890|
|ks_IN.UTF-8 | 1,234,567.890|
|tr_TR.UTF-8 | 1.234.567,890|

`grouping=False`:

| Locale | Formatted example |
|---|---|
|en_DK.UTF-8 | 1234567,890|
|en_US.UTF-8 | 1234567.890|
|fr_FR.UTF-8 | 1234567,890|
|ks_IN.UTF-8 | 1234567.890|
|tr_TR.UTF-8 | 1234567,890|

parsing using `locale.atof(s)`:

|locale | 1,234,567.890 | 1.234.567,890 | 1 234 567,890|
|---|---|---|---|
|en_DK.UTF-8 | error | 1234567.89 | error|
|en_US.UTF-8 | 1234567.89 | error | error|
|fr_FR.UTF-8 | error | error | 1234567.89|
|ks_IN.UTF-8 | 1234567.89 | error | error|
|tr_TR.UTF-8 | error | 1234567.89 | error|

One exception though: FreeCAD supports an expression language, i.e. `2 * 4` in certain input boxes. IMO for the sake of avoiding ambiguity, this format should only accept the "1234.56" number format, just like the vast majority of programming languages do. The fact that Excel is locale-aware in this sort of scenario has caused folks plenty of headaches; let's not bring those into FreeCAD.

flaviut, 2025-08-03:
Ok, another exception after doing more research: the sad fact is that most software does not have correct locale handling, so people are likely accustomed to always using the "1234567.890" style.

In that case, it is likely confusing to parse "1234567.890" as "1234567890.0"

RogerJL, 2025-11-03:
Trying FreeCAD 1.02 and I can not enter measurements correctly.
Actually adding an measurement and just accepting the shown value (i.e. writing back what was read) fails in every property combination I have tried...
Windows is English
Language and keyboard (has decimal , in numeric part) is Swedish
```
18:32:54  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 2.22 cm for the constraint with index 18 is invalid
18:32:54  Namnlös1#Sketch: App.getDocument('Unnamed1').getObject('Sketch').setDatum(18,App.Units.Quantity('22.200000 mm'))
```
or
```
18:40:34  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 11,12 mm for the constraint with index 18 is invalid
18:40:34  Namnlös1#Sketch: App.getDocument('Unnamed1').getObject('Sketch').setDatum(18,App.Units.Quantity('11.120000 mm'))

```
To me it looks like value is converted to decimal point, but then Units.Quantity expects decimal comma and fails

OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Swedish/Sweden (sv_SE) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion