FreeCAD/FreeCAD #22560: Addon Manager:  (42564 dev version) Unknown command 'Std_AddonMgr' message   when initializing FC

ErtugrulEge, 2025-07-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello everyone!
if it is duplicate I don't know. I searched with error messages github but I couldn't find any new issue so I decided to write. When I open FC add on manager gives error messages and Tools->addon manager is missing in this version.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-07-18:
Fixed
https://github.com/FreeCAD/FreeCAD/issues/22492

ErtugrulEge, 2025-07-18:
Thank you, so I am closing

closed 2025-07-18 by ErtugrulEge