FreeCAD/FreeCAD #22564: Sketcher - Some construction constraints with external geometry sometimes fail.

leoheck, 2025-07-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sometimes I cannot use the parallel constraint with the external geometry.

I cannot even use a Perpendicular constraint with the external geometry in this case.

Sometimes I cannot use the angular constraint to read the angle between 2 external geometries. 

Do you know if this is a bug? Why does this happen?

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * Assembly4 0.50.18
  * sheetmetal 0.7.24
  * DynamicData 2.74.0
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.2
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-07-18:
ah, I got a workaround, just placed a construction line over the external geometry so I could use all the constraints.

luzpaz, 2025-07-18:
Can you provide an example file, @leoheck ?

PaddleStroke, 2025-07-31:
Without the file it's hard to say, but I would guess that those external edges that give you trouble are not lines but flat bspline perhaps? Or do they show up as lines in the element list?

leoheck, 2025-08-15:
Sorry, forgot to comment here. Most of the time I cannot perform operations with the external geometry, I think this started to happen recently. I barely use bsplines, unless something convert lines to bsplines internally, they are always straight lines.

I am no sure of it is related but sometimes aplling a constraint is hard. I have to type de shortcut 2 or 3 times with the selected objects. But the issue above is not this one since the constraints fail with a annoying popup saying I cannot do what I want since the external like is not a valid line.

I still have to provide a sample file.

FEA-eng, 2025-10-16:
> I still have to provide a sample file.

Can you confirm if the issue is still relevant and/or share the example file?

FEA-eng, 2026-10-19:
Needs confirmation and example from @leoheck.