FreeCAD/FreeCAD #22570: FreeCAD versions 1.0.1 fail to import the Python cv2 module

Francisco-Rosa, 2025-07-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Until version 1.0, it was possible to use the internal Python module cv2 (opencv). It seems that starting with version 1.0.1, this is no longer possible. Workbenches like Movie, among others, use this module. It would be necessary to find out what's happening and fix it.

Thank you.

Below is the error message.

```
>>> import cv2
...
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/tmp/.mount_FreeCAFMggAJ/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAFMggAJ/usr/lib/python3.11/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/tmp/.mount_FreeCAFMggAJ/usr/lib/python3.11/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAFMggAJ/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAFMggAJ/usr/lib/python3.11/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/tmp/.mount_FreeCAFMggAJ/usr/lib/python3.11/site-packages/cv2/__init__.py", line 76, in bootstrap
    raise ImportError('ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.')
ImportError: ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.
```

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-library
  * Render 2024.12.15
  * Movie 2025.1.4
  * Solar 2025.5.8
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Here's the link to the issue reported on the forum: https://forum.freecad.org/viewtopic.php?t=98284

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-18:
@Francisco-Rosa is this also an issue on 1.1-dev ?

luzpaz, 2025-07-18:
Can reproduce on 1.1-dev

```python
>>> import cv2
Oops. An unexpected error occurred:
  File "<input>", line 1, in <module>
  File "/tmp/.mount_FreeCAekmbJm/usr/lib/python3.11/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/tmp/.mount_FreeCAekmbJm/usr/lib/python3.11/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAekmbJm/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAekmbJm/usr/lib/python3.11/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/tmp/.mount_FreeCAekmbJm/usr/lib/python3.11/site-packages/cv2/__init__.py", line 76, in bootstrap
    raise ImportError('ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.')

ImportError: ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.
```

```shell
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git) AppImage
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/149.623/1
Installed mods: 
  * FreeCAD-Ribbon 1.9.1.1
  * SearchBar 1.4.2
  * Silk 0.2.0
  * Telemetry 1.0.1
```

RCL-Loren, 2025-08-28:
I was looking into this a bit more, and might have uncovered something helpful. There seems to be, potentially, two issues. I am not set up to develop for FreeCAD, but just ran into this with the Movie workbench, and tried to find the issue. This might not be correct, but it seems right to me. I hope it can help a developer who is familiar with the codebase and build process.

1. Path Contamination issue: After some poking around in the python console, I found this cv2 path which doesn't seem like it should be there. 

`/Users/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/osx/APP/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/cv2/python-3.11`

It shows up in `/Applications/FreeCAD.app/Contents/Resources/lib/libopencv_core.4.11.0.dylib`

2. The shiboken2 binding generator replaces the python built-in __import__ function, and every import goes through the shiboken _import_ wrapper. This wrapper can modify import behavior for Qt related functions.

OpenCV's bootstrap process calls __import_ internally and causes a cycle like:

cv2.__init__.py calls bootstrap()
↓
bootstrap() tries to import binary extensions
↓
Import goes through shiboken2's _import wrapper
↓
shiboken2 calls original_import()
↓ 
Due to contaminated paths, import fails and retries
↓
Retry triggers another bootstrap() call
↓
RECURSION DETECTED!