FreeCAD/FreeCAD #22575: Crash in Part WB

RZHSSNZDH, 2025-07-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I click on some buttons, FreeCAD crashes and exits. I have seen this only in Part WB for example for fillet or chamfer. The error shown in terminal after crash is this:
```
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3def0) [0x7f790c04def0]
#1  0x7f7892fd04f7 in PartGui::DlgFilletEdges::onFilletStartRadiusValueChanged(Base::Quantity const&) from /usr/lib/freecad/lib/PartGui.so+0x137
#2  /usr/lib/libQt6Core.so.6(+0x1d37ef) [0x7f790c7d37ef]
#3  0x7f790ff21530 in Gui::QuantitySpinBox::valueChanged(Base::Quantity const&) from /usr/lib/freecad/lib/libFreeCADGui.so+0x40
#4  0x7f790ff22fc0 in Gui::QuantitySpinBox::updateFromCache(bool, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0xc0
#5  0x7f790ff23160 in Gui::QuantitySpinBox::updateText(Base::Quantity const&) from /usr/lib/freecad/lib/libFreeCADGui.so+0x70
#6  0x7f7892fe0bde in PartGui::DlgFilletEdges::DlgFilletEdges(PartGui::DlgFilletEdges::FilletType, Part::FilletBase*, QWidget*, QFlags<Qt::WindowType>) from /usr/lib/freecad/lib/PartGui.so+0x1ee
#7  0x7f7892fe21b9 in PartGui::TaskFilletEdges::TaskFilletEdges(Part::Fillet*) from /usr/lib/freecad/lib/PartGui.so+0x59
#8  0x7f7892f7768d in CmdPartFillet::activated(int) from /usr/lib/freecad/lib/PartGui.so+0x2d
#9  0x7f790f9360e7 in Gui::Command::_invoke(int, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0x3e7
#10  0x7f790f936647 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x147
#11  /usr/lib/libQt6Core.so.6(+0x1d37ef) [0x7f790c7d37ef]
#12  0x7f790d4073c0 in QAction::activate(QAction::ActionEvent) from /usr/lib/libQt6Gui.so.6+0x210
#13  /usr/lib/libQt6Widgets.so.6(+0x2424ad) [0x7f790da424ad]
#14  0x7f790da426c1 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0xf1
#15  0x7f790db85e61 in QToolButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0x31
#16  0x7f790d95b6c7 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xd87
#17  0x7f790d901c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x90
#18  0x7f790d905c39 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0xff9
#19  0x7f790f8c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#20  0x7f790c768118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#21  0x7f790d8f7398 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x5b8
#22  /usr/lib/libQt6Widgets.so.6(+0x176f13) [0x7f790d976f13]
#23  /usr/lib/libQt6Widgets.so.6(+0x1781d8) [0x7f790d9781d8]
#24  0x7f790d901c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x90
#25  0x7f790f8c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#26  0x7f790c768118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#27  0x7f790cfa1217 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x3c7
#28  0x7f790d0248f4 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x2f4
#29  0x7f790d013b27 in QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x197
#30  0x7f790c7c0e34 in QObject::event(QEvent*) from /usr/lib/libQt6Core.so.6+0x144
#31  0x7f790d901c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x90
#32  0x7f790f8c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#33  0x7f790c768118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#34  0x7f790c7684fb in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt6Core.so.6+0x38b
#35  /usr/lib/libQt6Core.so.6(+0x43fcf8) [0x7f790ca3fcf8]
#36  /usr/lib/libglib-2.0.so.0(+0x5e87d) [0x7f790adc787d]
#37  /usr/lib/libglib-2.0.so.0(+0x5fcd7) [0x7f790adc8cd7]
#38  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7f790adc8ee5]
#39  0x7f790ca3c5e2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x72
#40  0x7f790c7744b6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x206
#41  0x7f790c76c7c1 in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0xb1
#42  0x7f790f7c7bf5 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x9b5
#43  freecad(+0x8276) [0x55919a332276]
#44  /usr/lib/libc.so.6(+0x276b5) [0x7f790c0376b5]
#45  /usr/lib/libc.so.6(__libc_start_main+0x89) [0x7f790c037769]
#46  freecad(+0x8625) [0x55919a332625]
```
I have tested and seen this result also in AUR and flatpak version of FreeCAD.
I have the problem also in safe mode.
As the error, it is possible to be a problem in this line of the code:

https://github.com/FreeCAD/FreeCAD/blob/c968effe26b199864e6009c03a1f54d22cb89e3b/src/Mod/Part/Gui/DlgFilletEdges.cpp#L289C5-L289C35

Or this:

https://github.com/FreeCAD/FreeCAD/blob/c968effe26b199864e6009c03a1f54d22cb89e3b/src/Mod/Part/Gui/DlgFilletEdges.cpp#L928

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, OCC 7.9.1
Locale: Persian/Iran (fa_IR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

Part

### Anything else?

I have asked the question in the forum, but no useful answer!
https://forum.freecad.org/viewtopic.php?t=97860

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-18:
Can you reproduce on 1.1-dev ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

RZHSSNZDH, 2025-07-24:
I upgraded `freecad` via `pacman`. Now I can't even open a model! The app crashes and the error below appears:
```
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::140295163251648
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3def0) [0x7f9968e4def0]
#1  /usr/lib/libX11.so.6(XDefaultScreenOfDisplay+0x4) [0x7f9967ce6744]
#2  /usr/lib/libCoin.so.80(+0x461949) [0x7f9968861949]
#3  /usr/lib/libCoin.so.80(glxglue_init+0x43) [0x7f99688627d3]
#4  /usr/lib/libCoin.so.80(cc_glglue_instance+0x826) [0x7f99688635f6]
#5  0x7f99686b800d in SoGLRenderActionP::isDirectRendering(SoState const*) const from /usr/lib/libCoin.so.80+0x4d
#6  0x7f99686bd2c5 in SoGLRenderActionP::render(SoNode*) from /usr/lib/libCoin.so.80+0xc5
#7  0x7f99686bcedc in SoAction::apply(SoNode*) from /usr/lib/libCoin.so.80+0xcc
#8  0x7f996cabe33e in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x2e
#9  0x7f996cbdf87d in Gui::View3DInventorViewer::renderScene() from /usr/lib/freecad/lib/libFreeCADGui.so+0x17d
#10  0x7f996cb8c655 in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1d5
#11  0x7f996cb9b6fa in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x4a
#12  0x7f996a75b3f3 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xab3
#13  0x7f996a7c7a76 in QFrame::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x36
#14  0x7f99695675dc in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0xec
#15  0x7f996a701c60 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x80
#16  0x7f996c6c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#17  0x7f9969568118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#18  0x7f996a74c2de in QWidgetPrivate::sendPaintEvent(QRegion const&) from /usr/lib/libQt6Widgets.so.6+0x5e
#19  0x7f996b6f59a3 in QOpenGLWidget::event(QEvent*) from /usr/lib/libQt6OpenGLWidgets.so.6+0x1a3
#20  0x7f996a701c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x90
#21  0x7f996c6c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#22  0x7f9969568118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#23  0x7f996a757c4d in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x1cd
#24  0x7f996a758093 in QWidgetPrivate::setVisible(bool) from /usr/lib/libQt6Widgets.so.6+0x1c3
#25  0x7f996a7576c2 in QWidgetPrivate::showChildren(bool) from /usr/lib/libQt6Widgets.so.6+0x102
#26  0x7f996a757ad4 in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x54
#27  0x7f996a757955 in QWidgetPrivate::showChildren(bool) from /usr/lib/libQt6Widgets.so.6+0x395
#28  0x7f996a757ad4 in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x54
#29  0x7f996a758093 in QWidgetPrivate::setVisible(bool) from /usr/lib/libQt6Widgets.so.6+0x1c3
#30  0x7f996a7576c2 in QWidgetPrivate::showChildren(bool) from /usr/lib/libQt6Widgets.so.6+0x102
#31  0x7f996a757ad4 in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x54
#32  0x7f996a757955 in QWidgetPrivate::showChildren(bool) from /usr/lib/libQt6Widgets.so.6+0x395
#33  0x7f996a757ad4 in QWidgetPrivate::show_helper() from /usr/lib/libQt6Widgets.so.6+0x54
#34  0x7f996a758093 in QWidgetPrivate::setVisible(bool) from /usr/lib/libQt6Widgets.so.6+0x1c3
#35  0x7f996a900a11 in QMdiSubWindow::eventFilter(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x271
#36  0x7f99695675dc in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0xec
#37  0x7f996a701c60 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x80
#38  0x7f996c6c3eb9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#39  0x7f9969568118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#40  0x7f996a75815a in QWidgetPrivate::setVisible(bool) from /usr/lib/libQt6Widgets.so.6+0x28a
#41  0x7f996c64db9c in Gui::Document::createView(Base::Type const&) from /usr/lib/freecad/lib/libFreeCADGui.so+0x55c
#42  0x7f996c5c22c8 in Gui::Application::slotNewDocument(App::Document const&, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd08
#43  0x7f996baeb1aa in boost::signals2::detail::signal_impl<void (App::Document const&, bool), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::Document const&, bool)>, boost::function<void (boost::signals2::connection const&, App::Document const&, bool)>, boost::signals2::mutex>::operator()(App::Document const&, bool) from /usr/lib/freecad/lib/libFreeCADApp.so+0x2ba
#44  0x7f996bd7a9a4 in App::Application::newDocument(char const*, char const*, bool, bool) from /usr/lib/freecad/lib/libFreeCADApp.so+0x1924
#45  0x7f996bd7d465 in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&) from /usr/lib/freecad [... cut ...]
#95  /usr/lib/libc.so.6(+0x276b5) [0x7f9968e376b5]
#96  /usr/lib/libc.so.6(__libc_start_main+0x89) [0x7f9968e37769]
#97  freecad(+0x8625) [0x557d6a9f2625]
```
Now the version detail is this:
```
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.5, Qt 6.9.1, Coin 4.0.4, Vtk 9.5.0, OCC 7.9.1
Locale: Persian/Iran (fa_IR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

FEA-eng, 2026-09-01:
I assume that it's still relevant for the OP.

oxfordpelican, 2026-10-06:
I could reproduce this with FreeCAD 1.1.4 and a Persian system locale (fa_IR). It crashes when opening Part Fillet or Chamfer, with the same backtrace. With English or German locale it works.

With a build of current main it doesn't crash anymore, also with Persian locale. It was fixed by the QuantitySpinBox change in #32707, so weekly builds from 20 September on should be fine, and the next release too.

Workaround for 1.1: start FreeCAD with English number formatting, e.g. `LC_NUMERIC=en_US.UTF-8 freecad`.

RZHSSNZDH, 2026-10-06:
> I could reproduce this with FreeCAD 1.1.4 and a Persian system locale (fa_IR). It crashes when opening Part Fillet or Chamfer, with the same backtrace. With English or German locale it works.
>
> With [...]

Yes! It did be solved with layout changing to english in 1.1.4!