FreeCAD/FreeCAD #22580: weekly build: crash when attempting symmetric pad on a base feature created from stl

brettowe, 2025-07-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

on the attached file (compressed with 7zip as zip was i guess to big)

[grip-model.FCStd(7z).zip](https://github.com/user-attachments/files/21323777/grip-model.FCStd.7z.zip)

create a pad with the only sketch, model will disappear, then click symmetric (i think reverse also does it) freecad then crashes with the below see in console
if the original stl is required it was from here: https://www.thingiverse.com/thing:4614297

```
Program received signal SIGSEGV, Segmentation fault.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3def0) [0x7f677e860ef0]
#1  /tmp/.mount_freecagabOfp/usr/lib/././libTKBO.so.7.8(+0x23c18a) [0x7f6708e3c18a]
#0  /usr/lib/libc.so.6(+0x3def0) [0x7f677e860ef0]
#2  /tmp/.mount_freecagabOfp/usr/lib/././libTKBO.so.7.8(+0x23d857) [0x7f6708e3d857]
#1  /tmp/.mount_freecagabOfp/usr/lib/././libTKBO.so.7.8(+0x23c18a) [0x7f6708e3c18a]
#2  /tmp/.mount_freecagabOfp/usr/lib/././libTKBO.so.7.8(+0x23d857) [0x7f6708e3d857]
#2  /tmp/.mount_freecagabOfp/usr/lib/././libTKBO.so.7.8(+0x23d857) [0x7f6708e3d857]
#3  0x7f670c2a1db8 in OSD_ThreadPool::performJob(opencascade::handle<Standard_Failure>&, OSD_ThreadPool::JobInterface*, int) from /tmp/.mount_freecagabOfp/usr/lib/./libTKernel.so.7.8+0x58
#3  0x7f670c2a1db8 in OSD_ThreadPool::performJob(opencascade::handle<Standard_Failure>&, OSD_ThreadPool::JobInterface*, int) from /tmp/.mount_freecagabOfp/usr/lib/./libTKernel.so.7.8+0x58
#4  0x7f670c2a2232 in OSD_ThreadPool::EnumeratedThread::performThread() from /tmp/.mount_freecagabOfp/usr/lib/./libTKernel.so.7.8+0x92
#4  0x7f670c2a2232 in OSD_ThreadPool::EnumeratedThread::performThread() from /tmp/.mount_freecagabOfp/usr/lib/./libTKernel.so.7.8+0x92
#5  0x7f670c2a228a in OSD_ThreadPool::EnumeratedThread::runThread(void*) from /tmp/.mount_freecagabOfp/usr/lib/./libTKernel.so.7.8+0xa
#5  0x7f670c2a228a in OSD_ThreadPool::EnumeratedThread::runThread(void*) from /tmp/.mount_freecagabOfp/usr/lib/./libTKernel.so.7.8+0xa
#5  0x7f670c2a228a in OSD_ThreadPool::EnumeratedThread::runThread(void*) from /tmp/.mount_freecagabOfp/usr/lib/./libTKernel.so.7.8+0xa
#6  /usr/lib/libc.so.6(+0x957eb) [0x7f677e8b87eb]
#6  /usr/lib/libc.so.6(+0x957eb) [0x7f677e8b87eb]
#6  /usr/lib/libc.so.6(+0x957eb) [0x7f677e8b87eb]
#7  /usr/lib/libc.so.6(+0x11918c) [0x7f677e93c18c]
```

### Full version info

```shell
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git) AppImage
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: C/Default (C)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.8077/1
Installed mods: 
  * ThreadProfile 1.96.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-18:
Is this a regression ?

maxwxyz, 2025-07-19:
I can confirm the crash on macOS, in the weekly and also in the stable 1.0.1 so no regression.
@pieterhijma is this OCC related?

To be fair, a converted mesh model is not an ideal base for solids.

```
OS: macOS 13.7.6
Architecture: x86_64
Version: 1.1.0dev.42532 +7 (Git)
Build date: 2025/07/12 12:25:40
Build type: Unknown
Branch: part-design-transparent-previews
Hash: aec208ddb25d6c102a255d4c8d849486e60a1ee2
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 72/72/1
Installed mods: 
  * freecad.gears 1.3.0
  * CfdOF 1.31.4
  * Defeaturing 1.2.6
  * Gridfinity 0.11.9
  * AddonManager 2025.6.29
  * Manipulator 1.6.0
  * fasteners 0.5.38
  * fcVM.backup1752306371.4383721 (Disabled)
  * Telemetry 1.0.1
  * CurvedShapes 1.0.13
  * Cables 0.2.1
  * sheetmetal 0.7.24
  * FusedFilamentDesign 0.25.250
  * Curves 0.6.64

```

chennes, 2025-07-26:
Here's the backtrace of the main thread -- not much interesting happening here.
```
FCBRepAlgoAPI_BooleanOperation::Build(const Message_ProgressRange &) FCBRepAlgoAPI_BooleanOperation.cpp:111
Part::TopoShape::makeElementBoolean(const char *, const std::vector<…> &, const char *, double) TopoShapeExpansion.cpp:5791
PartDesign::FeatureExtrude::buildExtrusion(Base::Flags<…>) FeatureExtrude.cpp:796
PartDesign::Pad::execute() FeaturePad.cpp:74
App::DocumentObject::recompute() DocumentObject.cpp:168
Part::Feature::recompute() PartFeature.cpp:107
PartDesign::Feature::recompute() Feature.cpp:82
App::Document::_recomputeFeature(App::DocumentObject *) Document.cpp:2989
App::Document::recomputeFeature(App::DocumentObject *, bool) Document.cpp:3053
App::DocumentObject::recomputeFeature(bool) DocumentObject.cpp:204
PartDesignGui::TaskFeatureParameters::recomputeFeature() TaskFeatureParameters.cpp:75
PartDesignGui::TaskExtrudeParameters::tryRecomputeFeature() TaskExtrudeParameters.cpp:332
PartDesignGui::TaskExtrudeParameters::onMidplaneChanged(bool) TaskExtrudeParameters.cpp:941
<lambda>::operator()() const qobjectdefs_impl.h:152
QtPrivate::FunctorCallBase::call_internal<…>(void **, <lambda> &&) qobjectdefs_impl.h:65
QtPrivate::FunctorCall::call(void (PartDesignGui::TaskExtrudeParameters::*)(bool), PartDesignGui::TaskExtrudeParameters *, void **) qobjectdefs_impl.h:151
QtPrivate::FunctionPointer::call<…>(void (PartDesignGui::TaskExtrudeParameters::*)(bool), PartDesignGui::TaskExtrudeParameters *, void **) qobjectdefs_impl.h:199
QtPrivate::QCallableObject::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) qobjectdefs_impl.h:570
Gui::GUIApplication::notify(QObject *, QEvent *) GuiApplication.cpp:93
Gui::GUIApplication::notify(QObject *, QEvent *) GuiApplication.cpp:93
tryRunEventLoop(Gui::GUISingleApplication &) Application.cpp:2290
runEventLoop(Gui::GUISingleApplication &) Application.cpp:2319
Gui::Application::runApplication() Application.cpp:2406
main MainGui.cpp:321
```
Probably need a debug build of OpenCASCADE to make any more progress here, @pieterhijma .

GeeBee2, 2026-10-27:
With OCC 8.0.1 FreeCAD freezes and I see the following error.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 27.1.0dev.20261006 (Git shallow)
Build date: 2026/10/06 21:00:40
Build type: Release
Branch: (HEAD detached at 3424c4106)
Hash: 3424c410627263394f701aeec5b61d7bab7bb3a1
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 25.9.1.250822/ATI Technologies Inc./AMD Radeon RX 6900 XT
```

[image]

With OCC 8.1.0-dev it gives an error and doesn't crash or freeze.
```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.47994 (Git)
Build date: 2026/08/09 14:06:28
Build type: Release
Branch: (tags/weekly-2026.08.05-81-gba6f76ca9c-dirty)
Hash: ba6f76ca9ceb96e38400e69ec7d3658ed153ff08
Python 3.14.6, Qt 6.11.1, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.6.2, boost 1_91, Eigen3 5.0.1, PySide 6.11.1
shiboken 6.11.1, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.1.0.dev1-53706f78a8
Locale: German/Germany (de_DE) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 25.9.1.250822/ATI Technologies Inc./AMD Radeon RX 6900 XT

```

```
Failure while computing removed volume preview: Invalid input shape for boolean CMN:
Self-intersecting wire
Self-intersecting wire
Self-intersecting wire
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
Unorientable shape
No error
No error
```

pieterhijma, 2026-10-28:
It is clear that this model is very compute heavy for FreeCAD/OCCT because of the mesh that is converted to BRep. First, the shape is calculated by OCCT, then FreeCAD does its refine which runs forever as well. Modeling like this is something that FreeCAD/OCCT simply doesn't support. 

Of course, it shouldn't crash, but that is not the case anymore since OCCT 8.0.1. 

I verified not crashing with FreeCAD main with OCCT 8.1 dev:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 27.1.0dev.49321 +1 (Git)
Build date: 2026/10/27 12:27:31
Build type: Debug
Branch: adapt-to-occt-v8_1_0
Hash: 371a550ee620d900a88d2e96bf8cb9e47b7e45ec
Python 3.14.8, Qt 6.12.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.1, boost 1_92, Eigen3 5.0.1, PySide 6.12.0
shiboken 6.12.0, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 2.0.1, OCC 8.1.0.dev1-53706f78a8
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.5-arch1.1/AMD/AMD Radeon 780M Graphics (radeonsi, phoenix, ACO, DRM 3.64, 7.2.11-arch1-1)
```

I'm closing this issue.

closed 2026-10-28 by pieterhijma