FreeCAD/FreeCAD #22582: UI: Could not create pixmap from qss:images_classic\arrow-down-lightgray.png

BackwardEcho, 2025-07-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The "arrow-down-lightgray" is missing from the UI while the button functionality remains. FC seems to be looking for a png file, but only the svg file exists. Error report happens on startup.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/32.512/1.25
Installed mods: 
  * AddonManager 2025.7.17
  * fasteners 0.5.39
  * freecad.gears 1.3.0
  * FusedFilamentDesign 0.25.250
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-19:
@MisterMakerNL

maxwxyz, 2025-07-19:
FreeCAD crashed for me on Windows fir the first time opening and trying to switch themes to the new stylesheet themes:

[image]

Starting it after force quit gives more errors:

[image]

maxwxyz, 2025-07-19:
Changing themes in the first start setup gives different errors:

[image]

MisterMakerNL, 2025-07-19:
Arrows all seem to be there tho:

[image]

KurtMF, 2025-07-19:
I get this error too, with dev release:

> OS: Windows 11 build 26100
> Architecture: x86_64
> Version: 1.1.0dev.42605 (Git)
> Build date: 2025/07/18 16:20:04
> Build type: Release
> Branch: main
> Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
> Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
> shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
> Locale: English/United States (en_US)
> Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Turntable/Object center
> Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/
> Logical DPI/Physical DPI/Pixel Ratio: 96/20.32/3
> 

When I choose Freecad Dark style sheet in Preferences > Display > UI, arrows are missing in tree and toolbar.  Freecad Light is missing arrows in the tree but not the toolbar; and Freecad style sheet seems to find it's arrows OK.

BackwardEcho, 2025-07-19:
> aaaaaaaah damn cmake is only pulling the svg files, damnit.

Could that be why I am also missing the check marks in toggle settings? I don't get any error message about them, but I don't see any in my FC settings or elsewhere I used to see them.

MisterMakerNL, 2025-07-19:
Yes, but also wiping your user settings or enabling dark or light theme fixes this too.

furgo16, 2025-07-20:
Is this the same issue as https://github.com/FreeCAD/FreeCAD/issues/22476 ?

MisterMakerNL, 2025-07-22:
This should be fixed with https://github.com/FreeCAD/FreeCAD/pull/22597 which is merged already.

luzpaz, 2025-07-22:
@BackwardEcho can you confirm the fix when you get a chance ?

BackwardEcho, 2025-07-22:
> [@BackwardEcho](https://github.com/BackwardEcho) can you confirm the fix when you get a chance ?

[The dev build for Windows](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) didn't seem to update. I tried after MisterMakerNL said it was merged even though the build version did not change like the others. So I'll see in a few days when those builds get updated again or let me know if I should try from the non-Bundle version. Hopefully I won't have the separate problem I had before with the non-Bundle version, made it unusable for me.