FreeCAD/FreeCAD #22584: PartDesign: Pad to First on consecutive features does not work.

leoheck, 2025-07-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have this design where I place screw bosses on a complex and inclined wall.

For that, I have to use the Pad to First (face). But when I go to pad the second screw boss, I cannot pad it to the first (face) anymore. It grows to infinity, but if I use the Last, it works - exactly as padding to the First on the previous feature. Then, for the next screw bosses, I have to keep alternating Pad to First, Pad to Last.

Is this how this feature should work, or is this a bug? Thank you.

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * Assembly4 0.50.18
  * sheetmetal 0.7.24
  * DynamicData 2.74.0
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-07-19:
Now, making ribs, I could place 2 consecutive Pads to the First face.
But in the second feature, one rectangle grew to infinity while the others grew correctly to the First Face.

[image]

maxwxyz, 2025-07-19:
Please provide an example file.

FEA-eng, 2026-04-17:
@leoheck If it's still relevant, can you provide some example or at least more details about potential steps to reproduce (e.g. if it was the same (first) face).